arkypita/LaserGRBL #929: I am unable to print the whole picture of my test file.

Knight7171, 2020-03-22:
Hello everyone,

Just to inform you I am new to this but I am willing to learn. I am proficient in Chief Architect, Photoshop, Krita, learning Blender and Zbrush.

I purchased a Uttiny 6550 CNC laser engraver kit 10W on Amazon.
[https://www.amazon.com/gp/product/B0811TFNLD/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1](url)
I purchased this Uttiny 6550 CNC laser engraver kit 10W to cut 10 Mil acrylic sheets for stencils. I design my stencils in Krita using vectors and saving as PNG or JPEG files. I design all of my work in 3.00 Pixels for my lines. I am testing on cardboard first before I do it on the 60 sheets of acrylic that I purchased. I have not even opened up the boxes yet. 

My whole goal for this is to make my stencils. If I can do that then the second goal is to burn my artwork into wood. If I can do the first without issues then the second should be no problem. 

The other issue that I am having is how do I set a home home/return. Also my working area is 65x50 cm 25x19 inches. And one of my stencils I printed out in full size that I created and saved it in. Prints out smaller. BTW I tested out one of your files called Paul Smith. At 1000 power going by the settings that the machine came with it set the cardboard on fire. I caught it just in time and no damage to the laser. But on my circle test it makes little dots at full 1000 power and does not burn threw. I did notice that when I did my circle test that it did passes instead of drawing the circle. I would rather it draw then make left and right passes like the old dot matrix printers to make it.

I apologize for asking. I know that this is probably very simple for you all. I am new to all of this and using this software except for designing my stencils.

I have enclosed some pictures of my problems.

Any help would be greatly appreciated.

Thank You!

[image]
I did the same settings for both PNG and JPEG.
[image]
I did the same settings here as well for both PNG and JPEG.
[image]
These are the circles that I made in Krita from 3 pixels to 12 pixels. They are 2"x2" but print out in 7/16".
[image]
This one I did a centerline and it drew it out like I wanted. 30 sec's vs 10-13 min's.
[image]
The test file that came with LaserGRBL looks great and I think that if I did it multiple times it could cut through.
[image]
I did this one in both PNG and JPEG and it just makes dots. When I select Centerline or Vectorize it comes out blank.
[image]
This one drew it out in 30 seconds like I want all of mine to do. I have not yet tested it if it will cut with multiple passes but I am sure that it will.

StuartB4, 2020-03-22:
If they are not coming out the size you enter in the settings, then your X or Y or both axis steps/mm could be wrong. Also if you have GRBL 1.1 change the M3 in the Target image settings to M4 and make sure Laser Mode is enabled in GRBL configuration on the Grbl menu. 

You can calibrate the steps/mm with a spreadsheet and a ruler.  You can get it here:
[Calibration Spreadsheet.zip](https://github.com/arkypita/LaserGRBL/files/4366326/Calibration.Spreadsheet.zip)

Extract it and it should open in MS Excel or Libre Office.
Just follow the instrustions in the image that comes with it.

Knight7171, 2020-03-22:
Thank you StuartB4 for the quick reply. Ok I loaded up my test circle image. and where it says M3 in the Target image, I have no setting for M4 and I believe that Laser mode is enabled.
[image]
[image]

arkypita, 2020-03-22:
Laser mode is enable with 1

Knight7171, 2020-03-22:
Hello Arkypita, 
Ok I assume that you ment type in the gcode box M1. And the laser fired up quickly and before I knew it. I burn a hole right threw the cardboard. LOL!

Ok I think that I have my files figured out to print. Can you tell me if this looks right. And also how do I stop the dots in every circle that gets printed.

Thank You!

[image]
[image]
This one is the same file going from 3.0, 4.0, 5.0, 6.0, 7.0, 8.0, 9.0, 10.0, 11.0, 12.0. Pixels. And there is no difference in thickness from what I have made. Except for the size. Is this normal?

arkypita, 2020-03-23:
> Ok I assume that you ment type in the gcode box M1
No, I mean that you should set laser mode to 1 if you want to have laser mode active. You have zero in your configuration.

[image]

Keep in mind that when you say "your testfiles" I have no idea what you're talking about.

You bought an engraver, and they gave you a download. But the software they give you (LaserGRBL) is a free software that you can freely download from the internet. Those who sold you the engraver added the test files.

gmmanonymus111, 2020-03-23:
> I design my stencils in Krita using vectors...

FYI, LaserGRBL can process SVG files, if you can save it in that format, you don't need to load them as PNG/JPG and re-convert it to a vector. Also, LaserGRBL can't handle vector fill's, and will only draw the outline (see #873).

> This one is the same file going from 3.0, 4.0, 5.0, 6.0, 7.0, 8.0, 9.0, 10.0, 11.0, 12.0. Pixels. And there is no difference in thickness from what I have made. Except for the size. Is this normal?

AFAIK, vector lines represent 1 laser line, so the thickness is varied by your laser dot/rectangle (correct me if I'm wrong).

If you want to engrave a picture (and not just a vector) you can use line-to-line or dithering laser modes. These will engrave the whole picture (PNG/JPG) as seen on the preview. You might want to play with contrast, brightness and white clip to get the best results. Those modes will make thicker lines appear thicker, but the engraving takes more time (also, not suitable for cutting).

Knight7171, 2020-03-24:
Hello Arkypita and gmmanonymus111,

I am sorry for the delay in my reply. I have been dealing with other things as well.

Arkypita, Ok I understand now and got fixed. As far as the test files go. You are correct they added it to the download. And I made a file using vectors in Krita that is just test circles. Just to see how they turn out with different Pixel widths.

Gmmanonymus, Excellent to know. Unfortunately Krita cant save files as SVG. I will have to check out Eggbot. Thanks for the info. I tried the line-to-line mode on a picture and it would take around 4 hrs to complete...wow.

As I said I am new to this and I am learning as I go.

Now what about the pinhole dots in the circles? Is there a way to prevent it? And How would I setup a home/return spot on my machine?

Thank You!

gmmanonymus111, 2020-03-25:
> Now what about the pinhole dots in the circles? Is there a way to prevent it?

I think it should go away if you set laser ON command to `M4`. To do that, you have to enable laser mode by either sending `$32=1` command to GRBL or by modifying `$32` to `1` at `Grbl` -> `Grbl Configuration`

> And How would I setup a home/return spot on my machine?

You can set your zero position by clicking on the _Globe_ button. After every finished job, (or when you press the _house_ icon between the arrows) GRBL will return here. When you restart GRBL, this position (precisely, your work position) is lost, you need to set another zero position.
To do homing, you will need limit switches set up and wired to your board. Also, make sure that hard limits (`$21`) is enabled (set to 1). If you restart LaserGRBL with those set up correctly, You should see another _House_ button between the _Lightning_ and the _Lock_. Once presed, GRBL starts looking for the limit switches. When found, it'll set your zero position near there. This way, your zero position will be the same every time you do homing.

> And print orientation as well. My printer prints upside down from me on account of the size of the machine and the cable length

You can try playing around with `$3` (I think setting it to 2 will invert Y axis).
Note: to modify `${NUMBER}` settings, you have to be connected to your machine, and your machine has to be _unlocked_ and on _idle_ status.

Happy learning 😄

Knight7171, 2020-03-25:
Hello Gmmanonymus,

Thank you for the instructions. I am unable to do this now but I will be able to tomorrow. And learning something new is always fun until you run into lots of problems...LOL! I am sure that I will have more questions tomorrow. I will keep you up to date. And hopefully no problem. I love this program on account of what you can do with it. :)

Thank You!

Knight7171, 2020-03-26:
Ok. Nothing today is working like it should and I am getting so stressed and frustrated that I don't know what to do. A simple design does not come out right and I cant see this printing on the 10mm acrylic that I have already purchased.

I realized today that I need to resize my images in 12" x 12" and 12" x 17 1/2" because that is what my acrylic sheets are. And when I resize it. It ends up all jagged not clean. Ugh!

Any ideas anyone?

Thank you!

[image]
[image]

gmmanonymus111, 2020-03-26:
For these kind of vector images I suggest use  [InkScape](inkscape.org).
You can create vector (might be similar to krita), but you can save your image as SVG. Much easier to resize/edit the image, and LaserGRBL handles it better than centralised/vectorised PNG images. Preferably use v 0.91, as current stable 0.92 causes some problems with LaserGRBL when an object is rotated or resized.
No stressing, it'll work out good, it just needs a little more time and experimenting ^^

Knight7171, 2020-03-27:
Hi Gmmanonymus,

Ok, I downloaded 0.91 and used that to make a sample test file. and it does not print. and what about converting my JPEG or PNG images to SVG using Inkscape?

Very true but it does get frustrating.

Thank You!

gmmanonymus111, 2020-03-28:
Could you tell me more about 

> it does not print

?
Also, you can convert JPEG/PNG to SVG, but please note LaserGRBL currently can't handle SVG filling, as mentioned above.
I'll try to make some screenshots how I do SVG engraving.

gmmanonymus111, 2020-03-28:
Here are some screenshots
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Knight7171, 2020-03-28:
When I imported the SVG file. It would not print or anything. And I decided to try out the arrow buttons on LaserGRBL. And I don't think that it is jogging. If it is, then its to slow to really see it. 

Today I was able to print one of my PNG files. But I had to use Vectorize! But I noticed that when I print it. It prints the same line twice. once inside the line and once outside the line. I don't know how to fix it. Also for some reason my focal point was off so I adjusted it. I am not sure how that happened.

I like what you did and I see the problems that I made. In Inkscape I clicked on stroke to path instead of object to path and I did not use trace bitmap. I am still new to the program.

I also used a extension called Inkscape Laser Tool Plug-In from J-Tech. [https://jtechphotonics.com/?page_id=2012](url). And it worked somewhat.

gmmanonymus111, 2020-03-29:
> When I imported the SVG file. It would not print or anything. And I decided to try out the arrow buttons on LaserGRBL. And I finally got the jogging to work. But sometimes it sticks and will not move on the x axis.

This seems like your control board has difficulties keeping up with LaserGRBL's status reports. Try to set it `Fast `or `Slow` at `Grbl`->`Settings`->`Protocol`.
If it is only for X axis, please ignore the above reply.

> Today I was able to print one of my PNG files. But I had to use Vectorize! But I noticed that when I print it. It prints the same line twice.

It is the normal working mode of Vectorize. It'll always try to draw an inside and an outside line. To get only one, try Centerline.

> I also noticed today that when I try to increase my target image size W and H I can only go as high as 200mm 9" x 9" When I try to do 300 I get red boxes.

Make sure that `$110`, `$111`, `$112` are configured properly at `Grbl`->`Grbl Configuration`

Knight7171, 2020-03-29:
Ok I changed it and so far so good. Ok centerline works good.

Now how do these values look for my machine?

[image]
[image]

StuartB4, 2020-03-29:
> I also noticed today that when I try to increase my target image size W and H I can only go as high as 200mm 9" x 9" When I try to do 300 I get red boxes.

That's because the Max Travel is set to 200 in GRBL, change $130 and 131 to the maximum size your machine can engrave.

Knight7171, 2020-03-29:
Ok I see it. Got it. Thank you Gmmanonymus and Stuart. I am slowly getting the hang of it. LOL!

StuartB4, 2020-03-29:
Yeah, keep going, we all started at the beginning. :+1:

Knight7171, 2020-03-29:
So very true for all of us for everything in life.

Knight7171, 2020-03-29:
Ok I got it working somewhat. I changed it to 279.4 for it to print 11 inches it prints bigger. So for me to print 11 inches I need to set it to. 245.0. Its doable but very weird.

And how would I set up a home location somewhere in the middle of my work area? Example a 12"x17" sheet of acrylic in the middle of my work area that I can mark on my table so all of my stencils I can place them in the same spot even the 12"x12" and it will always start at the bottom right or left corner of the sheet.

gmmanonymus111, 2020-04-06:
Sorry for the late reply, last week was exams week, so I was a little busy.
For homing to middle you want to use offsets. I made a button to home to middle, but for this, you need to set up your machine's travel distances right. **Note: this config assumes you set your home location to bottom left corner of your machine.**
`$H (Home to Machine X0Y0Z0)`
`G10 L20 P1 X0 Y0 Z0 (Reset G54 to Mahine 0)`
`G90 X111 Y172 F2500 (Replace with your X an Y startup pos)`
`G10 L20 P1 X0 Y0 Z0 (Set offset, this will be your new zero)`
`G54 (Use offset)`

Now every job will use that offset as zero position.
**Also note: this was not tested on long-term run**

Knight7171, 2020-04-06:
Thank you Gmmanonymus, It's all good no worries I understand. 

So how would I go about setting my travel distance?

I copy and pasted the code and left out the parentheses. I have not tested it yet on account of the travel distance setting.

gmmanonymus111, 2020-04-10:
Travel distances should be measured by hand (I mean with tools, ofc). I measured mine by homing the machine to zero (either corner of the whole workplace, mine was lower left), then I mounted a pen to draw a line, and went up as far as my machine let me (when getting closer to the end, I set a lower distance count, not to break anything), and when it stucked, I repeated the process by going right 'til the end (got something like ┌), and measured my 2 lines (225*347 mm). By this, I decided the max travel values should be 221 mm for X and 343 mm for Y (leaving a little out on both ends not to ram the machine to the end).
The best option would be knowing the exact sizes by the manufacturer, which is not quite possible if it is a home-made machine.

Knight7171, 2020-04-19:
Thank You Gmmanonymus. I think that I have it set and so far so good. Time will tell. I was fiddling around with it before and when it hit the stoppers it was still clicking and I had to reduce the number. many times. lol! but I think that I have it set at the right distance.

closed 2020-12-16 by arkypita