arkypita/LaserGRBL #930: LaserGRBL Power Settings Messed Up

chasehensley, 2020-03-23:
Hi all, I got Laser GRBL and when I want to set the laser to the lowest power I have to do S1000 not S1 and when I do S1 the laser turns on. This means that I can't get any GCODE to run because they all have their S#'s under 1000. Has anybody ever had this happen to them?

StuartB4, 2020-03-23:
Do you mean when you enter S1 it turns on at full power?
What machine/laser do you have, and have you used other software that worked correctly?

chasehensley, 2020-03-23:
No sorry, I mean that when I do S1 nothing happens the way I can get the laser to turn on at the lowest power is to do S1000. I have a 3018prover.

StuartB4, 2020-03-24:
Have you enabled Laser mode in GRBL configuration settings?

StuartB4, 2020-03-24:
One other thing, don't have the offline controller connected at the same time as the USB connection.
They don't like it apparently.

chasehensley, 2020-03-24:
Ok, Ill try unplugging the Offline Controller. But how do I turn on laser mode? Is it just $32 = 1?

StuartB4, 2020-03-24:
Yes. Make sure you are connected, goto Grbl menu, Grbl Configuration. Change $32 to 1 and make sure $30 is 255 or 1000. $31 should be 0. Then click Write and exit.

taz101, 2020-03-24:
Hi, New to the laser, so I am having power problems. When I open GRBL the line numbers jump from $27 to $100. There is no $$30 or $31 can someone tell me what I am doing wrong. The laser comes on but does not burn.

StuartB4, 2020-03-24:
You need to have GRBL 1.1.
Start LaserGRBL and click Connect and it will tell you which version you have in the top of the window on the left.
[image]

StuartB4, 2020-03-24:
You can update it using LaserGRBL.
Start it but don't click connect but make sure the USB cable is plugged in.
Goto the tools menu and click Flash Grbl Firmware.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

taz101, 2020-03-24:
Thank you will try this later today
Paul

[1]<https://user-images.githubusercontent.com/16980026/77416713-24637b00-6dbc-11ea-9c2a-2a7b181e76b3.jpg>

[2]<https://user-images.githubusercontent.com/16980026/77416738-2c231f80-6dbc-11ea-9156-0a07178f1420.jpg>

[3]<https://user-images.githubusercontent.com/16980026/77416750-2fb6a680-6dbc-11ea-9fd9-0a34c53ca5e5.jpg>

[4]<https://user-images.githubusercontent.com/16980026/77417596-81136580-6dbd-11ea-8ce7-0d1c2b7d5918.jpg>

[5]<https://user-images.githubusercontent.com/16980026/77417647-8ffa1800-6dbd-11ea-9cbb-0f513a50b6cf.jpg>

[6]<https://user-images.githubusercontent.com/16980026/77416771-3b09d200-6dbc-11ea-8851-2f4dbd18d4c0.jpg>

[7]<https://user-images.githubusercontent.com/16980026/77416780-3e04c280-6dbc-11ea-908f-6fe1c6473644.jpg>

chasehensley, 2020-03-24:
Thanks! Is the Arduino Nano the correct one for a Sainsmart 3018prover? Thanks for all of the great information!

StuartB4, 2020-03-24:
@chasehensley 
Have a look at your machine controller board for the Atmel chip, the square one with the legs on all 4 sides.
It should be an Atmel 328P.

closed 2020-12-16 by arkypita