arkypita/LaserGRBL #931: Sainsmart 3 axis controller w/ laser control no grbl v1.1 functions

l7artlife, 2020-03-24:
Help please, 

I am running your software and the cnc movement functions work. It shows grbl 0.9 on the status line. The laser will not start or fire. I have tried everything and am at my wits end. 

Any recommendations? I have read that I need grbl v1.1 to gain access to the laser control functions. However the firmware tool is grayed out and will not let me use it. 

Thank you 
Larry

l7artlife, 2020-03-24:
3/24/2020 3:06:18 PM	Program     	------- LaserGRBL v3.1.2 [64bit] START -------
3/24/2020 3:06:20 PM	SetStatus   	Machine status [Disconnected]
3/24/2020 3:06:30 PM	SetStatus   	Machine status [Connecting]
3/24/2020 3:06:30 PM	OpenCom     	Open COM11 @ 115200 baud DTR, RTS, Ctrl-X
3/24/2020 3:06:30 PM	SendImmediate	Send Immediate Command [0x18]
3/24/2020 3:06:31 PM	VersionInfo 	Detected Grbl v0.9j
3/24/2020 3:06:31 PM	SetStatus   	Machine status [Idle]
3/24/2020 3:06:37 PM	SetStatus   	Machine status [Run]
3/24/2020 3:06:40 PM	SetStatus   	Machine status [Idle]

l7artlife, 2020-03-24:
00062	00612976	tx	$$\n
00063	00612980	rx	$0=3 (step pulse, usec)\r
00064	00612983	rx	$1=25 (step idle delay, msec)\r
00065	00612986	rx	$2=0 (step port invert mask:00000000)\r
00066	00612991	rx	$3=3 (dir port invert mask:00000011)\r
00067	00612993	rx	$4=0 (step enable invert, bool)\r
00068	00612996	rx	$5=0 (limit pins invert, bool)\r
00069	00612999	rx	$6=0 (probe pin invert, bool)\r
00070	00613002	rx	$10=3 (status report mask:00000011)\r
00071	00613004	rx	$11=0.010 (junction deviation, mm)\r
00072	00613007	rx	$12=0.002 (arc tolerance, mm)\r
00073	00613010	rx	$13=0 (report inches, bool)\r
00074	00613013	rx	$20=0 (soft limits, bool)\r
00075	00613013	rx	$21=0 (hard limits, bool)\r
00076	00613015	rx	$22=0 (homing cycle, bool)\r
00077	00613021	rx	$23=0 (homing dir invert mask:00000000)\r
00078	00613023	rx	$24=25.000 (homing feed, mm/min)\r
00079	00613026	rx	$25=500.000 (homing seek, mm/min)\r
00080	00613029	rx	$26=250 (homing debounce, msec)\r
00081	00613031	tx	[3F]
00082	00613032	rx	$27=1.000 (homing pull-off, mm)\r
00083	00613034	rx	$100=400.000 (x, step/mm)\r
00084	00613037	rx	$101=400.000 (y, step/mm)\r
00085	00613038	rx	$102=400.000 (z, step/mm)\r
00086	00613042	rx	$110=1600.000 (x max rate, mm/min)\r
00087	00613045	rx	$111=1600.000 (y max rate, mm/min)\r
00088	00613048	rx	$112=1600.000 (z max rate, mm/min)\r
00089	00613051	rx	$120=10.000 (x accel, mm/sec^2)\r
00090	00613054	rx	$121=10.000 (y accel, mm/sec^2)\r
00091	00613056	rx	$122=10.000 (z accel, mm/sec^2)\r
00092	00613059	rx	$130=200.000 (x max travel, mm)\r
00093	00613062	rx	$131=200.000 (y max travel, mm)\r
00094	00613065	rx	$132=200.000 (z max travel, mm)\r
00095	00613065	rx	ok\r

arkypita, 2020-03-24:
> the firmware tool is grayed out

The firmware tool is grayed out because you can use it only when disconnected. If you connect to the machine, you cannot flash firmware.

arkypita, 2020-03-24:
If the laser does not turn on it's not a software problem or grbl version problem.
Grbl 1.1 has a much more better support for laser, but 0.9 should be able to turn on laser too.

Check your connections. https://github.com/arkypita/LaserGRBL/wiki/Laser-Wiring

closed 2020-12-16 by arkypita