arkypita/LaserGRBL #932: Adding option to Export w/ Header and Footer

RelentlessInnovator, 2020-03-26:
I was exporting code to my offline laser controller, and I ran into problems because the exported code did not have a header. I understand the reasoning for this not being the default from #783, but I think having a separate option to export with a header would be very helpful instead of having to manually add a header/footer every time you export code for those of us who prefer to use offline controllers.

arkypita, 2020-03-27:
Hello @RelentlessInnovator 
I really appreciated the fact that you went to investigate the topic looking if it had already been discussed.

Yes, I approve of what you are saying and I think adding this option is quite simple, so I will implement it in the next release.

I think that I will create a form that ask what to export, including the code between passes and number of passes (duplicating the code multiple times). This could be useful to people like you that want to use the generated gcode with another streamer.

RelentlessInnovator, 2020-03-27:
Thanks @arkypita ! I look forward to seeing this feature implemented!

Overall, I am very impressed with this program! It has been great for running my new laser cutter. Thanks for making this program, and sharing it with the community!

arkypita, 2020-03-30:
https://github.com/arkypita/LaserGRBL/releases/tag/v3.1.3