arkypita/LaserGRBL #933: cut the  the external contour first.

gabrieldan69, 2020-03-28:
Hello all,

I use a cnc 3018 pro + 10w laser
I have the following question: how can i cut the external line first ?
I mean to cut the  the external contour first.

want to  cut a shape and engrave in the center of it without cutting it in the center.

Best regards

gabrieldan69, 2020-03-28:
[image]

gmmanonymus111, 2020-03-28:
I think it is only possible with 2 seperate files. One 'engrave' file with internal contour and another 'cut' file with the external contour. LaserGRBL can't recognise if the line is an inside or outside contour.
I would suggest to engrave it first, and then cut it, otherwise your engraving might get shifted a little bit.

DrGio018, 2020-05-06:
You need to edit your G-code, by lowering the S value for engraving area.
Or, as gmmanonymus111 wrote, make two separate files for cutting and engraving, and join them. In this case, you will have to add a small object (i.e. 1mm square) at the bottom left corner of the picture, and use it in both files as a zero-point.