arkypita/LaserGRBL #935: Changes in color themes

GusanoGris, 2020-03-28:
I have allowed myself to modify the application's colorization module so that custom themes can be added through a json file.

This time I have cloned the project through VS 2019 but, although it seems that it manages the versioning, I just did not see how to send it to you.

As an example I send you a couple of screenshots with the application working.

I'm sorry for my awkwardness with GitHub, but I've almost always worked on unshared projects and when it hasn't, I've used other project management tools.

Also, my English is lousy.

[image]
[image]

GusanoGris, 2020-04-01:
The code is already incorporated in the version I'm working with and I've added the use of hex color to make coloring more flexible.

I include the modified files attached to the message.

You will also see other changes that I am adding, but these will take longer because my intention is to be able to use the color of the SVG as a laser power modifier {M (n) P (m)} and that allows me to burn the SVG with different levels.

For now it seems that it generates the gcode correctly, although it uses both M3 and M4 and I don't know if that will work).

What I can't quite get is to plot the intensities in the viewing fields.

I hope you find it interesting. A greeting.
[cambios.zip](https://github.com/arkypita/LaserGRBL/files/4415197/cambios.zip)

arkypita, 2020-04-01:
Hi @GusanoGris 
Any contribution to development is very welcome. I understand that you are not familiar with github but you would do well to learn how to use it to make the integration of changes a simple and controlled thing.

You should download a git manager, such as "github desktop". https://desktop.github.com

Create your own fork of LaserGRBL project, work on it, and then pull the changes to main project.

You can ask help to @falias4 or @gmmanonymus111 or @guillaume-rico that sent their code contribution via github pull

arkypita, 2020-04-01:
The package you sent me conflicts with the changes I made in the last few days, and with the commit that other people added.

[image]

It is not easy to merge it manually, this is why a collaborative versioning system is needed when you want to collaborate to a distributed project.

GusanoGris, 2020-04-01:
Sorry for the problem, I am seeing how to use github with VisualStudio.

I am advancing in the process

a greeting

arkypita, 2020-04-01:
In any case, I would prefer to have an internal color scheme editor in LaserGRBL (like with a form that edit color scheme) rather than relying on external configuration files. Finally, I'd rather avoid incorporating external .dll libraries like the json one because I like the code to be completely available as a source.

So even if I really appreciate the work you have done, I don't think I will want to insert it as it is within LaserGRBL.

arkypita, 2020-04-01:
Unfortunately I'm not a great github expert either, I started using it for the LaserGRBL project (I use SVN for work) so I can't give you many indications about how to use it correctly.

GusanoGris, 2020-04-01:
Ok, in this case I will leave the topic parked.

Regarding third-party libraries, it is a completely respectable opinion and, although I have always been in favor of not reinventing the wheel more than once, I understand that you prefer to have control of the code.

Thanks for your attention.

arkypita, 2020-04-01:
Just to be precise: I am not against the reuse of third-party code, without which LaserGRBL would not even exist, but I prefer the inclusion of sources over the use of external .dlls, where possible.