arkypita/LaserGRBL #938: Z axis end stop

wfishrn, 2020-03-29:
Only running 2 axis, X and Y, grbl set to home ZYX in that order, so homing fails, any ideas on how to disable the Z homing or can I some how do a work around, Im even thinking of using a momentary switch if needed to fool the machine into thinking there is a Z axes??
Thank You

gmmanonymus111, 2020-03-29:
You need to recompile GRBL to disable Z homing AFAIK.
To do so, edit config.h in GRBL's folder:
Replace
```
#define HOMING_CYCLE_0 (1<<Z_AXIS)
#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))
```
with
```
#define HOMING_CYCLE_0 (1<<X_AXIS)
#define HOMING_CYCLE_1 (1<<Y_AXIS)
```
to do X homing first, and Y next, or
```
#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))
```
to do X and Y homing at the same time

wfishrn, 2020-03-29:
Ok will this affect the laser connected to the Z+ for PWM control, just thought of that.. ???

StuartB4, 2020-03-29:
No. I assume you are using a Protoneer CNC shield and UNO.
I do it the same way. X,Y home, pwm wire for laser connected to Z+/- on shield.
Works fine.

closed 2020-12-16 by arkypita