arkypita/LaserGRBL #939: HELP - LaserGRBL with Ender 3 Printer

darkworks, 2020-03-29:
Hi  GRBL Overloads :) i need some help with this tool.

I have Ender 3 printer with Silent Mainboard V1.1.5 which mean i can load GRBL firmware into it.
[image]
[image]

I have bought 20 w laser module which support pwm: https://www.aliexpress.com/item/4000001195993.html

so my plan is to stick this laser module with printer extruder through magents and take connection from extruder fan to the pwm pin of laser module , and using dc-dc buck converter  drop the 24 v to 12 v of ender power supply for laser module. 

Problem : 
How can i tell LaserGRBL that which motor of my printer is x , y  and which are the limit switches ? so anyone who can guide me on hardware configuration , i think it will be good project to make LaserGRBL working on Ender 3 printer :) 

thanks

StuartB4, 2020-03-29:
Not sure about the work needed to make it work for you, never tried fitting on to a 3D printer.
You would probably get a lot more help about the hardware side of it on Facebook groups from people who have done that sort of thing. Just one other thing, you do know that the laser you bought uses 20 watts of power but is only a 5.5 watt output. Have a look here: http://nejetool.com/modules.html

StuartB4, 2020-03-29:
As for Marlin, this is from the LaserGRBL website page. 

> Marlin firmware support is in a very early stage. It is assumed that LaserGRBL is able to communicate with Marlin, but the gcode generated by LaserGRBL could be not 100% compatible.

darkworks, 2020-03-30:
> Not sure about the work needed to make it work for you, never tried fitting on to a 3D printer.
> You would probably get a lot more help about the hardware side of it on Facebook groups from people [...]

oh i see :( so they sold me  5.5w laser as 20 w , will open dispute

StuartB4, 2020-03-30:
They did'nt con you, so a dispute may not work.
If you look on the listing you posted of the laser it says 5.5 output power.

[image]

darkworks, 2020-03-31:
hmmm actually i purchased it from this seller : https://www.aliexpress.com/item/4000214959199.html?spm=a2g0s.9042311.0.0.27424c4dKJGO7m

there is no Optical Power mentioned in his description that's why my eyes not catch it that its 5.5 w laser

StuartB4, 2020-03-31:
Ok. Take a screenshot of their page and contact them to see what they say. You can always send them the screenshot and show them.

closed 2020-12-16 by arkypita