arkypita/LaserGRBL #941: $3 = 5 (Step direction invert, mask)

quotelawrence, 2020-03-30:
after using the modifications to the settings $3=7, and the lettering and directions 
are all correct if I hit the home button the machine now travels so far 
in the wrong direction I get a travel arrest and the machine shuts down
is there another spot where I need to change a setting to stop the 
machine from going to a position that does not exist, ?

gmmanonymus111, 2020-03-30:
[GRBL homing setup](https://github.com/gnea/grbl/wiki/Set-up-the-Homing-Cycle#homing-direction) might be helpful for you

quotelawrence, 2020-03-30:
ok thanks I am still very ignorant of the different functions and setting so I noted i used $3 = 7, and the table took off in the y direction opposite as if the zero (home was in the negative area, does that make any sense I will experiment I just hate crashing the machine over my ignorance

closed 2020-12-16 by arkypita