arkypita/LaserGRBL #942: Arduino with not connect Laser GRBL

mustafapat, 2020-04-01:
Hello,

First of all, thank you for your support for this beautiful project.

I'm having a problem.
Arduino can't connect the LaserGRBL program with the motherboard.
I connect with BenBox but why can't I connect with LaserGRBL.

I couldn't find the right firmament.I tried them all.

gmmanonymus111, 2020-04-01:
To my knowledge, BenBox uses a custom firmware for their machines. You could try flashing GRBL 1.1, but I'm not sure about your machine's original pinout.

arkypita, 2020-04-01:
I confirm what @gmmanonymus111  say
LaserGRBL work with machine that use grbl as firmware, Benbox firmware machine is not supported.

It is possible to flash grbl firmware to the controller board. Obviously you will lose the possibility of using the Benbox program (but I don't think it's a great loss) in exchange you will be able to use more advanced software designed for grbl.

However, since it is a modification to the machine firmware, it cannot be guaranteed that it is risk-free. The biggest issue is that you probably don't have the original firmware of the machine, so if you have problems it would be impossible to go back. The second issue is that after flashing you will have to configure parameters that you probably don't know the values of, which however can be established experimentally.

Get informed and search on google and youtube, where you can find tutorials about "benbox to grbl".

For flashing grbl firmware it is possible to use the flashing tool available inside LaserGRBL under tools menu.

mustafapat, 2020-04-01:
Thank you for your valuable reply. I want to use your program with this kit
I have.

closed 2020-12-16 by arkypita