arkypita/LaserGRBL #944: Laser Burning while Traversing White Space

CssTechGuy, 2020-04-05:
Hello,

I have a SainSmart 3018 CNC that came equipped with a Woodpecker 3.2 Control Board.
The spindle function works great and I have been able to mill some good projects.
I ordered the SainSmart 5500mw laser to broaden my horizons.
Unfortunately, the laser is ruining my attempted projects as it is burning while traversing through the white space.
I am running GRBL V1.1h and no matter what adjustments I make the issue is not resolved.
Any help would be greatly appreciated.
Attached are my GRBL settings.

[image]

StuartB4, 2020-04-05:
Don't you have to switch from Spindle mode to laser mode somehow on the control board.
I've seen it mentioned somewhere.

CssTechGuy, 2020-04-05:
Stuart,

Yes I've seen that myself on certain boards with jumpers.
There are no jumpers on my Woodpecker 3.2 control board.
There is only a push on/off switch button on the Laser Control Board.

DrGio018, 2020-05-06:
Could be a problem in G-code with M3/M4.

closed 2020-12-16 by arkypita