arkypita/LaserGRBL #946: Help on customizing the firmware - feed hold with a

gisep, 2020-04-05:
Hello guys,
more than one time i tought to experiment some customization on the firmware, as controlling a light source or, as in this case, to add a thermostat to auto-pause the job with temperature control.

My question is: how can i customize the firmware?
I can write some basics in arduino, i'm not an expert, but it's ok with taking my time and studying the necessary to do something.

I can hardly understand that the LaserGRBL code (and others too) is done by assembling "blocks", so you can have a hierarchical and synthetic seen to get oriented and operate for sectors in specific subfiles.
My goal is to **change the "feed hold" and resume buttons to a push button**, since i've got a thermostat (reading a probe in the laser case) with a switch that automatically goes on when the temperature limit is overpassed, and returns off when it returns cooler.

1) How can i understand the structure of LaserGRBL firmare in order to know how to modify it?

And...

2) how many free inputs-outputs, digital or analogic, are there in the ArduinoUNO with cnc shield running the last LaserGRBL firmware?

p.s. the thermostat i'm using is called W1209.

I will be glad for any answer.

 Giuseppe

arkypita, 2020-04-05:
LaserGRBL is not a firmware, is a software.
Grbl is the firmware and the developer site is here: https://github.com/gnea/grbl

Ciao

arkypita, 2020-04-05:
For what I know  grbl use all the program and memory space of atmega328 chip, also it is very very compact and optimized code so it is difficult to modify.

I suggest to use a separate arduino to perform temperature control, and interface it to grbl feed hold fisical pin to pause/resume job

gisep, 2020-04-06:
Thank you Arkypita,
I confused the software LaserGRBL with the proper firmware to run it (grbl). Sorry!
I will write an analogue post there.

>  grbl use all the program and memory space of atmega328 chip [...] so it is difficult to modify.

True, but i think that this change won't add something more, but only modify a switch from a "toggle" to a "momentary" one. Moreover, it would free one pin! (the resume one)

Maybe it could be reasonable to try the WeMos D1 instead of the Arduino UNO because of the more ROM (in addition to the integrated WiFi). The problems could be its (not tested) stability and the only lonely analogic port.

I'll let you know!
Bye

closed 2020-04-07 by arkypita