arkypita/LaserGRBL #949: Laser on and off issue

nickm324, 2020-04-09:
I cannot get LaserGrbl to run on or off my laser. Movements controls work fine and it works fine with other laser applications like LaserWeb without issue. Where do I need to start to resolve this issue. I am running Grbl 1.1g, have laser mode set.

When sending m4 nothing happens.

StuartB4, 2020-04-09:
Do you have PWM mode enabled in LaserGRBL settings?
[image]

nickm324, 2020-04-09:
Yes, it is working now. Had to send M3 then M4 a couple of times for some reason then it started working even after I had already restarted my controller a few times. Not sure what it was.

closed 2020-04-09 by nickm324

StuartB4, 2020-04-10:
Ok Nick, that's good. :+1: