arkypita/LaserGRBL #951: Laser CNC 3018 GRBL 1.1 not burning

Urimart, 2020-04-11:
Hello, 

I have an Issue that CNC 3018 laser is no longer burning. The board is a Developer 2.0 Arduino Nano that came with GRBL 0.9 and I have recently updated to 1.1. 

For the updated I used X loader selecting device (Nano ATmeta 328) and a hex grbl_v1.1h.20190825.  I checked in arduino IDE and the update was succefull. 

The spindle worked but the the laser is no longer burning. When I start the program withLaserGRBL it only turns on the blue light but not doing engraving. I have tried different settings with ($31, $32 comands) but still not work.  I see comands sent in console and are only M3 and M5 with out S value. 

I have the laser conected to Hightlight yellow Pin. 

[image]

thanks in advance of any help

StuartB4, 2020-04-11:
What does it say on the board in front of the yellow socket, can't see it very well?

Do you have PWM enabled in LaserGRBL Settings on the Grbl menu? On the Raster Import, second tab at the top.

Looking at the left side of the board, the black transistor above the with 12v+ socket is where the 3 wire laser would normally fit. If it says + - and s on the board you could probably connect the power wires from the laser (red and black) to the little white socket and then solder the PWM wire from the laser to the S pin of the transistor.

StuartB4, 2020-04-11:
Just found a better picture of the board.

[image]

What you can do if you have a multimeter/test meter is switch it on to continuity/beeper and then put 1 probe on the D11 pin of the blue Nano board and the other on the 3 pins in turn on the yellow socket. If the meter beeps when you touch one of them that will be the PWM pin that needs to connect to the PWM wire on the laser. Not all sockets are wired the same, so the pins may be in the wrong order for your laser. You just need to find out which is PWM, and which is Ground, the other will be the 12v +.

Singman33, 2020-05-22:
Try M3 S250 (if your laser support 255 levels of intensity) or M3 S1000 (if 1000 lvl) and do move command.

closed 2020-12-16 by arkypita