arkypita/LaserGRBL #952: Laser cutter not connecting

eskurt, 2020-04-12:
When I try to press the green play button to start cutting, it's not green because the device is disconnected. It is plugged in correctly and everything has been installed. I try to connect it and it tries for a sec and then still says disconnected. How can I fix this? Thanks!

StuartB4, 2020-04-12:
What machine do you have? Did you have to install a USB driver.
If your main controller board has a Nano board plugged in to it you
might need to install the CH340 driver. You can do that in LaserGRBL, on the tools menu.
Also check in Windows device manager to see if the Com port for the machine is there. 

If it says it's connected you can also try changing the Threading mode in LaserGRBL Settings for a slower one.

[image]

[image]

eskurt, 2020-04-12:
Hi, the laser cutter is CNC2-3040 and I’m using it on my laptop which has windows 10. I already downloaded and installed the driver, and checked the device manager. It was there, but when I go to LaserGRBL it still says disconnected. I changed it to the correct baud rate and tried, but still no luck. Will changing the threading mode affect whether I can connect? Thanks!

eskurt, 2020-04-12:
I tried changing the threading to all the possible slower options and it still isn’t commecting

arkypita, 2020-04-12:
Are you sure that CNC2-3040 controller use grbl as firmware?
LaserGRBL can connect only with grbl machines.

http://lasergrbl.com/faq/

eskurt, 2020-04-13:
The laser cutter came with a file to download LaserGRBL so I assumed so, but I’ll try what the website says. If you can come up with anything else, please let me know, thank you!!

arkypita, 2020-04-13:
> The laser cutter came with a file to download LaserGRBL so I assumed so
Ok, let's assume that it use grbl.

> I tried changing the threading to all the possible slower options and it still isn’t commecting
Thread speed doesn't have much to do with the connection problem. It may have some effect, but it actually affects most of the problems you might have with low quality arduino clones on communication stability, after connection.

It could be that your serial port needs to be "reset" on connection. Try to configure "hard reset" on the connection.

[image]

arkypita, 2020-04-13:
If it does not work read this.

Unlike other programs, LaserGRBL waits for grbl (the software that runs on the machine) to send it a correct status message, before considering itself validly connected.

The message grbl should send is something like this:
<Idle|MPos:0.000,0.000,0.000|FS:0.0,0> for grbl v1.1

If LaserGRBL does not receive this message in 10 seconds, it consider itself not connected (or connected to a non grbl device).

You can log the serial session and upload here, so I can see what is the problem.

1) Run LaserGRBL
2) Click on ? menu, select "activate extended log"
3) Select a filename where to record extended log
4) Click on connect button

This will log the communication during connection phase. The format of a valid log look like this:

```
00000	00352092	log	Recording session started @ 13/04/2020 10:54:11
00001	00354790	com	Open COM5 @ 115200 baud Ctrl-X
00002	00354800	tx	[18]
00003	00354801	tx	[3F]
00004	00354829	rx	\r
00005	00354830	rx	Grbl 1.1h ['$' for help]\r
00006	00354838	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:S|WCO:0.000,0.000,0.000>\r
00007	00355003	tx	[3F]
00008	00355010	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:S|Ov:100,100,100>\r
00009	00355203	tx	[3F]
00010	00355209	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:S>\r
00011	00355402	tx	[3F]
00012	00355408	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:S>\r
00013	00355604	tx	[3F]
00014	00355609	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:S>\r
00015	00355802	tx	[3F]
00016	00355807	rx	<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:S>\r
[...]
```

eskurt, 2020-04-17:
Hi,

I did that and it still won’t connect to my computer

StuartB4, 2020-04-17:
Have a look here, there may be something to help.
https://github.com/teachables/Laserfiles/tree/master/CNC2-3040

gmmanonymus111, 2020-04-17:
You can also try to connect to other software, like [Universal GCode Sender](https://winder.github.io/ugs_website/) to verify if your engraver is equipped with GRBL.

eskurt, 2020-04-23:
Hi,

I tried all of the suggestions but nothing works, so I ordered an offline controller. When I press the buttons to position or move the laser, it doesn’t move. The laser doesn’t turn on either. I double checked every connection and still nothing.

eskurt, 2020-04-23:
The laser is now on, but still now moving.

StuartB4, 2020-04-23:
Are you using the offline controller with the USB still connected. If so try unpluggin the USB first.

eskurt, 2020-05-06:
No I had it disconnected

gmmanonymus111, 2020-05-06:
Just to make sure you have GRBL installed properly on your arduino:
Connect your boad to your pc.
Please download [Putty](https://www.putty.org/).
Once downloaded, run it.
In the following window, set connection type to serial, then type in your arduino board properties. (Port/serial line, baud rate/speed)
[image]
You can find the com port the same way StuartB4 did it in the first reply. Baud rate should be `115200 `or `9600`
Click `Open`
You should see a similar text once connected successfully: 
[image]
This means GRBL is properly installed, and you can communicate with it. By sending `?`, you should get a response like this: `<Alarm|WPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:HS>`
If this is the case, LaserGRBL is at fault.

After connecting, you get something like this:
[image]
It means the baud rate is not correct.

In case you can not connect, you most likely have a wrong driver/USB cable.

closed 2020-12-16 by arkypita