arkypita/LaserGRBL #954: Wavy laser engraved lines

teamtexas55, 2020-04-14:
I'm just purchased an  Ortur 15 watt laser and have downloaded GBRL and updated the firm ware to 1.34 version.  When I run an engraving file that has curved lines I'm getting a wavy line instead of a smooth curve.  It doesn't happen everywhere though  Dan
[image]

gmmanonymus111, 2020-04-15:
My suggestion in the first place is to check if every screw is tight enough. Loose screws might cause issues like this.

teamtexas55, 2020-04-15:
Thanks Alex.  I did check both drive bands and adjusted them. I will double check for loose screws.  I tested a few more times and have found the wavy lines always come back in the exact same spots. Makes me think it’s a program glitch.

gmmanonymus111, 2020-04-15:
Which mode do you use to create the engrave-image in LaserGRBL? Also, could you upload the original image?

teamtexas55, 2020-04-15:
Here are some screen shots of the settings I used and the original image in jpeg.

[image]
[image]
[image]
[image]

gmmanonymus111, 2020-04-15:
For me, it produces good results, not as wavy as yours.
[image]
Excuse the poor wood quality

- The first one is made with `Smooth (HQ Bicubic)`, this looks way better.
- The second one is made with `Sharp (NearestNeighbor)` like yours, if you look closely, there is a small wavyness(?) in it, but not that noticable compared to yours.

You should give it a try with `Smooth (HQ Bicubic)` to see if it gets better.
What version of LaserGRBL and GRBL are you using? I can't quite read it, because your images are too small, and I don't recall GRBL has 1.34 version (only 0.9 or 1.1)

teamtexas55, 2020-04-15:
It does look better on the 1st one.  The version is GBRL 3.1.2.  I will give it a try on Smooth and see what it does.

[wine]<https://user-images.githubusercontent.com/9658243/79361715-f9e18980-7f45-11ea-8736-dd2c836fa8d9.png>

teamtexas55, 2020-04-15:
I ran the file again after I updated the firm ware on the laser to 1.34 version.  In smooth this is my result. 
[image]

This is reduced in size a bit, but seems much smoother than it did before I updated the firmware.  This was run at 3000 mm/sec speed and 750 Power.   I left the brightness, contrast slides at the default.

gmmanonymus111, 2020-04-16:
Looks better. Could you post your GRBL settings (by sending `$$` command to GRBL)?
You could also try with lower speed, as belts tend to stretch on high load.

arkypita, 2020-04-16:
I think it has too high acceleration and maximum speed values, so when the head moves quickly for jumping empty spaces the machine and the motors cannot do it with sufficient precision.

I am talking about this parameters

[image]

Check yours, write them down somewere (or export to file) and try set lower values.

teamtexas55, 2020-04-19:
Here are the settings. 
[image]
[image]
[image]

What do you think?

gmmanonymus111, 2020-04-19:
Sorry @teamtexas55, but your images are too small to read.

teamtexas55, 2020-04-19:
Lets try that again.
[image]
[image]
[image]

gmmanonymus111, 2020-04-19:
Looks much better now!
Everything seems fine, except X & Y acceleration (`$120`, `$121`), which seems too high for me.
My machine has 25 mm/sec^2, but it is a leadscrew-driven one, so your belt driven one can go faster obviosuly, although `2200` and `1800 `is a bit much. Try to lower it to `1320 `& `1080 `(60%) and check if it gets any better. Lower again, if needed.
Maybe someone with more experience with belt-driven engravers could suggest better values.

Junction deviation (`$11`) is set to `0.010` for me. This may also affect curves.

My step pulse time (`$0`) and step idle delay (`$1`) are half of yours, but it should not cause any problems the way it engraves.

closed 2020-12-16 by arkypita