arkypita/LaserGRBL #955: Wrong position during vector laser cut

mdeel-online, 2020-04-15:
Hello, at first... sorry for my bad english

I have a new toy. A Totem Laser engraver and i have one Problem only at vector laser cut.
I hope you can see the problem and know what i mean.
The laser does not move to the right place during cutting. It just starts in the middle. I don't have the problem with normal operations.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

mdeel-online, 2020-04-16:
I have found the cause and here it is explained in detail. Please note that the first G0 code must remain on G0. Otherwise the switched-on laser moves to the starting point

https://github.com/arkypita/LaserGRBL/issues/53

Now the question is to activate the old commands in the Lasergrbl?

[image]
[image]

filap, 2020-04-17:
It looks like I have the same problem ... but why only few people face this ... as everyone use the same LaserGRBL. and arduino load ?

StuartB4, 2020-04-17:
@mdeel-online. 

I noticed in the images you posted of you GRBL configuration you don't have Laser mode enabled ($32). In grbl menu, grbl configurating change it to $32=1. 
You can read what it does here: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode

Also $120 and 121 seem a bit high for Acceleration. If it's too fast it can lose steps.

mdeel-online, 2020-04-18:
Perfect thank you thats the Problem. 120 and 121 works fine at value 2000 and 32 its on now. Great Help from you!!!

arkypita, 2020-04-18:
that's what I tell you here #53

closed 2020-04-18 by arkypita