arkypita/LaserGRBL #956: Not returning to correct home position after job

MarkDowes, 2020-04-15:
Hi all, total noob here with a homing question.

At the beginning of a job I am pressing the "set zero point" button, i load my g-code, press go, it does its thing all in correct position, but when it is done it returns to a different point somewhere between where it ended and the position it started in.
Then, when I press the home button, it is attempting to return to home but overshooting by a long shot and forcing itself out of bounds and then the motor/belt are skipping.

What am I doing wrong? I have looked through the issues but I am not understanding.

Cheers.

MarkDowes, 2020-04-15:
Trying again, the job cut but then it tried to return home at the end of the job and it overshot right away, didn't return to where it started, afraid at this rate I am going to damage my belts and motors!

arkypita, 2020-04-16:
First question: what codes do you have in your settings-gcode form? In footer you should have something like:

`G0 X0 Y0 Z0`

[image]

This is necessary to return the laser to zero-zero at the end of the work. You can see this line added at the end of the processing.

[image]

Second question: what are the X/Y numbers you see in the the top right of preview immediately after pressing the "Set Zero Position" button (just a moment before starting the job)?

First line is the global position of the machine (machine position). When you push the "set zero" button a second line will appare with current offset around this new zero point (work position).

And how much are these numbers worth once the work is finished?
If they have the same values as the beginning, LaserGRBL did everything correctly and your laser should be at the same point as start.

MarkDowes, 2020-05-10:
Hi just to report back on this, after a lot of playing around, I think that my issue was that I had the maximum rate too high, it must have been skipping steps and ending up in the wrong place. This is the best guess I have anyway, it seems to be sorted now.

Thanks for your help and your continued development of this software, very much appreciated.

closed 2020-05-10 by MarkDowes