arkypita/LaserGRBL #957: Laser Cutting Delay To Full Power

outlawjack99, 2020-04-16:
Hi All,

My laser has a delay when trying to cut an image where it starts to cut around 9 seconds after the motion has started and it doesn't complete the sections of the image. I'm using a three part M character for testing and it misses the start and doesn't complete a complete section. I don't know if this is to do with the settings I use when "import raster image", on the main screen or something deeper.

[image]

arkypita, 2020-04-16:
9 seconds? its a lot of time!

Does the laser not turn on for the first 9 seconds, or does it turn on but doesn't cut? What material do you want to cut? What colour?

outlawjack99, 2020-04-17:
It seems a bit random in its choices. The laser always comes on, starts moving then it goes sharper and starts to cut. The new pic shows the order in which it cut each section and for 1 it started immediately but didn't finish, 2 it went through most of the motion before it started cutting and 3 was more or less the same. I don't think the imported image is at fault unless I'm setting the image up wrong?

[image]

gmmanonymus111, 2020-04-17:
While cutting white paper, you need to delay your movement a bit for the laser to actually start cutting. Once it burned the paper, it can continue & finish cutting. White paper reflect laser light, and caused similar issues with my 1.5w laser.

outlawjack99, 2020-04-17:
Well I'll be. I tried it on some brown then red paper and it worked a treat. So I take it the white stops the laser from "biting"? Either way it seems a thank you is in order.

gmmanonymus111, 2020-04-17:
> So I take it the white stops the laser from "biting"?

Exactly.
I'm glad it worked out.

closed 2020-12-16 by arkypita