arkypita/LaserGRBL #958: Wrong dimensions with Centerline Mode, if "Adaptive Quality" checked

tomturican, 2020-04-16:
Hello, thank you for this super good program, which offers me a lot of help. I am very pleased that you are so busy working on it and doing it better and better. The new "Adaptive Quality" function is very interesting and helpful. But unfortunately a problem has arisen. I very often use the "Centerline" function to cut out smaller parts for model making. However, if the checkbox of the new "Adaptive Quality" function is activated, the wrong size images are imported in Centerline mode. I have 3 screenshots here to show what I mean. The image size determination using EXIF works and shows the correct values, but at the last step the error seems to occur.

[image]
Adaptive Quality is unchecked. All is OK.

[image]
Now i check the Adaptive Quality and then switch to Centerline.

[image]
Now already in Centerline Mode. The new imported image got the wrong size.

arkypita, 2020-04-16:
Thanks for the quick feedback, I verify and release a correction as soon as possible.

closed 2020-04-16 by arkypita

arkypita, 2020-04-24:
Fixed in v3.3.0

commit 4236d1fd (2020-04-16): Fix #958