arkypita/LaserGRBL #962: Laser doesn’t come on for .nc files

LJg81, 2020-04-17:
Hi. First time to LaserGRBL, first time on github. 
When I open a raster image in LaserGRBL, the laser comes on and cuts/engraves as per the settings in the target image dialog; no problem.

However, when I load an .nc file, I’m not given that dialog to set power/speed of laser, and when I Run program, the motors move the laser around the drawing, but the laser does not activate.

Typing M3 S50 also turns the laser on for focussing.

Can anyone advise what I need to do to set the laser power and speed and to have it come on when running a .nc file?

Thank you.
Lee

StuartB4, 2020-04-17:
Open the .nc file with a text editor, something like Notepad++ which is free and gives you line numbers, unlike notepad in Windows. See if there are any Laser on options in the code, like M3 or M4.
What did you use to generate the gcode file?

LJg81, 2020-04-17:
Hi @StuartB4 thank you, I’ve downloaded Notepad++ and no, there isn’t any M3 or M4 mention in the code.

The G-code I have created in Scan2CAD

- the file is one that others have created and shared for making medical visor headbands.
- I downloaded the .svg version of this and tried to open in LaserGRBL, and it gives me a small Target image dialog allowing me to set speed and power; when I hit run it moves, but the laser doesn’t come on.
- so I downloaded the .dxg version instead, opened it in Scan2CAD, and saved it out as .nc Gcode, which is where I currently find myself. 

Is it as simple as adding in ‘M3’ at the start of this .nc file as I read it in Notepad++ and then ‘M5’ at the end, re-saving from Notepad++ and then re-opening that file in LaserGRBL?

StuartB4, 2020-04-17:
What you can do is load an image and prepare it for burning but instead of clicking start click the File menu and save the file. You should get a gcode .nc file for that image. Open that with a text editor.
At the top it should tell you the Power  S and Speed F settings and the laser on you set in Target image settings, and also the Laser on command M3 or M4, depending on what they were set to in Target image settings.

[image]

LJg81, 2020-04-17:
This is really so helpful @StuartB4, thank you. I appreciate how frustrating it must be having to deal with these elementary questions.

Thank you though, I now have the laser coming on. I can play around with speed and power from here I think.

Three more primary-school questions from me..

1. When I open an .nc file, I don’t get a target image dialog, is that correct? No matter, the laser is coming on so I think that’s all I’d need it for.

2. Line $30 of my grbl config is 1000 ($31 is 0). Am I therefore correct in assuming if I have S500 say in the gcode, that is telling to operate at 50% power?

3. How do I set it the start position of the laser relative to the image in the laserGRBL main window? 
What I’ve been doing is:
- the blue crosshair is at 0:0.
- I’m just manually pulling my laser to the far bottom corner of my laser cutter bed.
- then I sometimes I seem able to hit the globe icon next to the padlock where the custom buttons are, but currently it is greyed out.
^ in theory, if I can get that custom button active again, is that how I would be best advised to ensure the laser ‘starts’ it’s programme from the correct reference point?

I will post the first few lines of code if that helps. 

Thank you again(!)

LJg81, 2020-04-17:
This is what I have for the first 20 or so lines of my gcode: @StuartB4 

%

N10 G92 X 0 Y 0 Z 0
M3 S500
F1000

N20 G90

N30 G00 Z 0
N40 G00 X 453.06939 Y 107.77087
N50 G01 Z -0
N60 G01 X 437.01418 Y 112.07285 Z -0

N70 G00 Z 0
N80 G00 X 437.01418 Y 112.07285
N90 G01 Z -0
N100 G01 X 437.01418 Y 112.07285Z -0
N110 G03 X 436.23772 Y 112.17508 I -0.77645714 J -2.8977775

N120 G00 Z 0

LJg81, 2020-04-17:
And, hopefully my last question, the laser seems to be travelling too far. The framework of the laser is (X) 540mm x (Y) 470mm, and the design is 497 in total wide.

However, with the laser starting at the bottom left of the X/Y axis when I hit Run, it follows that blue line up, which should only be about 440mm along the X axis, and yet it overshoots this and grinds against the end of the X axis framework.

Am I reading the measurements on this screen incorrectly or is this scaling error coming from the gcode somehow?

[image]

arkypita, 2020-04-17:
Download this file and run on your engraver: [TEST GCODE.zip](https://github.com/arkypita/LaserGRBL/files/4494166/TEST.GCODE.zip)

[image]

This should draw a 50x50 mm square.
If your result have different size, go to  menu  "Grbl - grbl configuration" and set $100 and $101 using this formula.

$100 = $100 x 50 / width
$101 = $101 x 50 / height

StuartB4, 2020-04-17:
Q1. Yes, because all the settings for the file are included when the gcode is generated from any program, there is no need to set anything else. That's fine, but some programs generate different commands in some of the code that not all programs understand. Like for cnc machines you might have a tool change command, when you try to use that file in a laser engraver it does'nt understand the extra commands.

Q2. Yes half the S Max number whether it's 255, 1000, 10,000 should correspond to half the power, and a quater a quater power etc, etc.

Q3. My home position is bottom left so I've never used the Globe button or re-positiond the laser anywhere else. I know the button is only active when you've move the blue cross/laser away from the home position.

LJg81, 2020-04-17:
@arkypita that was precisely it - thank you! It cut that initially at 100mm x 100mm and so I have been able to correct that now thank you.

@StuartB4 Thank you for all your help. All up and running now. If we weren’t in lockdown I would buy you a beer or 6. 

Finally, at the end of 3 frustrating days, you’ve sorted it in the best part of an hour!

closed 2020-04-17 by LJg81

LJg81, 2020-04-19:
Follow up question. Have completed about 20mins of laser cutting on full power. Was about halfway through the design. Now laser appears to have lost power, and is now just a feint blue dot. Now worried I should not have been operating it at 100% power. 
Have I just killed the laser do you think?

arkypita, 2020-04-19:
It is probable. If you have bought a laser overpowered, underdissipated, and sold for more powerful than it really is, by ebay or aliexpress. It's an incorrect and very very unpleasant practice which unfortunately is widespread. Read this; https://github.com/arkypita/LaserGRBL/issues/780

I'm sorry and I can imagine the disappointment, complain to who sold you the laser and pretend a refund.

LJg81, 2020-04-19:
Hi @arkypita thank you. I suspect you are right.

The laser was only bought for this purpose for a little community group in our village to try and produce some PPE for our local doctors trust. Unfortunately we didn’t have much of a budget.

A classic case of you get what you pay for I guess. I can’t even imagine how I post this back to them - it took about 3hrs to put together this week.. 

I suppose better laser modules are available that I could fit to it? Though that would involve more outlay I guess.

You are correct in your assumption; a unbranded Chinese model bought via amazon.

arkypita, 2020-04-19:
I'm sorry, unfortunately there are many unfair people.
Where are you from?

LJg81, 2020-04-19:
Just outside Nottingham, England. Everyone on here has been very helpful. Thank you for that. I’ll try and return the laser. Amazon have a good returns policy I guess. I just won’t dismantle it fully.

arkypita, 2020-04-19:
I'm sorry, I don't know vendors in your area.

I bought my laser from the united states, from [dtr-laser](https://sites.google.com/site/dtrlpf/) only diode and heat sink were about $ 100 per 2W of power.

I know of a good seller from russia. A little expensive for my taste, but the product is exactly what it says. https://endurancelasers.com

I suggest you make the return with amazon, give a replacement, and then use the LaserGRBL settings to switch between work cycles and cooling cycles (you will find an entry in the configuration).

LJg81, 2020-04-19:
Thank you that is very helpful