arkypita/LaserGRBL #963: Noob with Stepper Motor Problems

JCowper, 2020-04-19:
Hello,

I am not having any success getting the Stepper motors to move with the Laser GRBL program. Any suggestions would be appreciated. 

I have also taken a picture of my wiring job to the CNC Shield on my Arduino Uno.

[image]

StuartB4, 2020-04-20:
You missed a pin on the motor connectors on the cnc shield. There's one left empty. Move the plugs down 1 pin.

JCowper, 2020-04-20:
Thanks, but no change to the problem.

arkypita, 2020-04-20:
Are you sure of the orientation of the stepper drivers? Mine have a serigraphy that says where the EN pin is, so I have no difficulty, but I don't see any indication on yours.

gmmanonymus111, 2020-04-20:
If that is a drv8825 driver, it is most likely been put together all right. (the EN tag is printed on both the shield and the unederside of the driver)

Could you check that blue 12V input socket on your shield? On mine, the soldering broke off when I tightened the screws. It is barely visible, but when you try to turn the screw, the little pin (on the downside of the shield) turns as well. If that's the case, you only need to resolder it.

Schwarzi07, 2020-05-20:
I had the same problem, but adjusting the current limiter Potentiometer made the problem go away. Have you also checked that you are using the correct voltage for your stepper motor

closed 2020-12-16 by arkypita