arkypita/LaserGRBL #964: My cutting lines doenst fit

son-y, 2020-04-19:
I don't know if its a software or a hardware problem. But my cutting lines doesnt fit. You can see the problem in the pictures below. The interruptions are not the problem because they depend on material and speed.

any ideas what i can try to fix it?

[image]
[image]

my grbl config is:
$0=10 (Step pulse time)
$1=25 (Step idle delay)
$2=0 (Step pulse invert)
$3=6 (Step direction invert)
$4=0 (Invert step enable pin)
$5=0 (Invert limit pins)
$6=0 (Invert probe pin)
$10=1 (Status report options)
$11=0.010 (Junction deviation)
$12=0.010 (Arc tolerance)
$13=0 (Report in inches)
$20=0 (Soft limits enable)
$21=0 (Hard limits enable)
$22=0 (Homing cycle enable)
$23=0 (Homing direction invert)
$24=25.000 (Homing locate feed rate)
$25=1500.000 (Homing search seek rate)
$26=250 (Homing switch debounce delay)
$27=1.000 (Homing switch pull-off distance)
$30=1000 (Maximum spindle speed)
$31=0 (Minimum spindle speed)
$32=1 (Laser-mode enable)
$100=80.000 (X-axis travel resolution)
$101=80.000 (Y-axis travel resolution)
$102=80.000 (Z-axis travel resolution)
$110=2000.000 (X-axis maximum rate)
$111=2000.000 (Y-axis maximum rate)
$112=2000.000 (Z-axis maximum rate)
$120=40.000 (X-axis acceleration)
$121=40.000 (Y-axis acceleration)
$122=40.000 (Z-axis acceleration)
$130=170.000 (X-axis maximum travel)
$131=180.000 (Y-axis maximum travel)
$132=150.000 (Z-axis maximum travel)

harryberlin, 2020-04-19:
i think, check your hardware. looks like unstable connection between motor and axle.

son-y, 2020-04-19:
All screws are fixed. And the belt is quite tight. What Else can I check? The Homing works really well, so the laser ends where it has started.

harryberlin, 2020-04-19:
in my case the x direction was not fixed correctly. had some differences.
one direction was fine. on invert (back) the direction, i have seen a small gap after some millimeters.

son-y, 2020-04-19:
And how did you fixed it? What did you change?

son-y, 2020-04-19:
And if you look on the pictures above you can See for example the Letter "h" wich is to short (y) and to thin (x) and not in proportion.

I'm really confused

arkypita, 2020-04-19:
Are you sure that grbl is correctly configured?

Download this file and run on your engraver: [TEST GCODE.zip](https://github.com/arkypita/LaserGRBL/files/4494166/TEST.GCODE.zip)

[image]

This should draw a 50x50 mm square.
If your result have different size, go to  menu  "Grbl - grbl configuration" and set $100 and $101 using this formula.

```
$100 = $100 x 50 / width
$101 = $101 x 50 / height
```

arkypita, 2020-04-19:
Your configuration look good.
In my opinion it's hardware problem.

Belts may also be too hard, giving the opposite effect. Or there may be friction on the axles or motors may not have enough current to win friction.

son-y, 2020-04-19:
I played with the belt thightness, seems to be a little better - I think I will try to find the best solution and will laser your calibration data!

Thanks a lot for the support!

closed 2020-12-16 by arkypita