arkypita/LaserGRBL #965: problema con la versione 3.3.0

monk91, 2020-04-19:
dopo l'aggiornamento ho riscontrato difficoltà a impostare correttamente le dimensioni del file SVG, inoltre dopo aver caricato un immagine jpg direttamente sul programma la mia stampante mi ha stampato il lavoro con un effetto specchio, ossia al contrario con le scritte che si leggevano al contrario, com'è possibile risolvere?

arkypita, 2020-04-19:
Ho fatto diverse modifiche che dovrebbero aver migliorato proprio la capacità di leggere i file SVG.
Ho provato con diversi file presi qua e la da internet e non ho avuto problemi.

Purtroppo SVG è uno standard aperto, e si trovano diverse implementazioni. Sto facendo fatica a stare dietro alle tante varianti.

Puoi temporaneamente scaricare la precedente versione 3.1.2 da qui: https://github.com/arkypita/LaserGRBL/releases/tag/v3.1.2

Mandami i file che ti danno problemi, e qualche spiegazione di quali effetti hai.
Puoi inviarli a info@lasergrbl.com

Per il fatto che è venuta invertita non dovrebbe centrare nulla con l'aggiornamento.
Magari domani mi spieghi meglio quest'ultima cosa ma per me dipende dalla configurazione degli assi di grbl. Non hai toccato nulla per sbaglio?

monk91, 2020-04-19:
No assolutamente, ho solo effettuato l'aggiornamento e provato a fare un incisione, ho interrotto la stampa a metà quando andando a controllare mi sono accorto che stampava ad effetto specchio...con un immagine non è un problema, ma con le scritte è un grosso problema xD

inoltre sempre dopo l'aggiornamento mi legge male le dimensioni dei file SVG portando immagini di 5cm a circa 3 volte più grandi, quindi non saprei più come impostarle...

arkypita, 2020-04-19:
> No assolutamente, ho solo effettuato l'aggiornamento e provato a fare un incisione, ho interrotto la stampa a metà quando andando a controllare mi sono accorto che stampava ad effetto specchio...con un immagine non è un problema, ma con le scritte è un grosso problema xD

Devi invertire gli assi, non è una cosa controllata da LaserGRBL ma da grbl, il software che gira sulla scheda di controllo. Ha alcuni parametri di configurazione, tra cui il $3, che serve appunto per invertire le direzioni degli assi.

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#3--direction-port-invert-mask

Setting Value | Mask | Invert X | Invert Y | Invert Z
-- | -- | -- | -- | --
0 | 00000000 | N | N | N
1 | 00000001 | Y | N | N
2 | 00000010 | N | Y | N
3 | 00000011 | Y | Y | N
4 | 00000100 | N | N | Y
5 | 00000101 | Y | N | Y
6 | 00000110 | N | Y | Y
7 | 00000111 | Y | Y | Y

arkypita, 2020-04-19:
mandami anche questi svg và, poi domani ci guardo

monk91, 2020-04-19:
[Nuovo Archivio WinRAR ZIP.zip](https://github.com/arkypita/LaserGRBL/files/4500398/Nuovo.Archivio.WinRAR.ZIP.zip)

gmmanonymus111, 2020-04-20:
Your SVG-s appear to be bigger when opened in LaserGRBL as is. When you open it in Inkscape, click `File`->`Document properties`, and set units to milimeters. 
[image]
Save your SVG image.
LaserGRBL's SVG library can not process SVG files in other format right now.

arkypita, 2020-04-20:
Confermo che il problema delle SVG non dipende dall'aver aggiornato alla versione 3.3.0 ma al non aver impostato il file in mm, la versione precedente 3.1.2 si comportava esattamente allo stesso modo. Segui le indicazioni di @gmmanonymus111  che funziona.

Per il futuro vedrò se si riesce a interpretare correttamente anche con le altre misure. Vedremo.
Per l'orientamento degli assi ti ho risposto.

arkypita, 2020-04-24:
@gmmanonymus111 
@monk91 

I have released a new version that fix the issue:
https://github.com/arkypita/LaserGRBL/releases/tag/v3.3.1

Alexandermolinagonzalez, 2020-04-24:
Exelente aplicacion , la uso en mi cnc laser de co2,
quisiera saber como la programo para hacer gravado y corte al mismo tiempo

closed 2020-12-16 by arkypita