arkypita/LaserGRBL #967: No G0 X0 Y0 at the end of gcode.

StuartB4, 2020-04-20:
I don't think it's me. I noticed in the latest version 3.3.0 when engraving finishes the laser just stops at the end of the last line, switches off and does'nt return home.

Tried it on 2 seperate computers with same results.
When I looked at the saved gcode I noticed there was no G0 X0 Y0 at the bottom.

arkypita, 2020-04-20:
Hi @StuartB4 
Starting from pre-release 3.0.19 (official release v3.0.24) i have moved starting gcode, ending gcode and gcode between passes, under user control in settings.

Can you check what values you have here?

[image]

arkypita, 2020-04-20:
I tried to make this change in the most compatible way possible, automatically setting the default values (both for new installation and updates) as you can see in the picture. 

Are you sure you haven't changed them by mistake?
It could be that they got lost with the update, but it would be very strange because in case of corrupt or missing settings I use the default (G0 X0 Y0 Z0).

StuartB4, 2020-04-20:
Oh right. That's ok. I did remove them from the settings menu options and assumed they were still added to the gcode when it was generated because they are at the beginning.

closed 2020-04-20 by arkypita