arkypita/LaserGRBL #968: One Test Run

theblankist, 2020-04-22:
I have the Ortur Laser Master 15w laser. Just set it up. Got everything up and running. Managed to run a decent test engraving. Suddenly the laser will not work on it's lowest setting. I can get it to project the laser on a medium or high (weak) setting, but it stops after 100 seconds. When I try to run anything there is no laser projection. 

[image]
 
I have no idea what happened or how to fix this.

theblankist, 2020-04-22:
There is no problem with stepping or moving the machine. The laser simply does not work. I ran an entire project with no problems. Tried to run it again with lower power, and now the laser won't work at all.

arkypita, 2020-05-06:
Messages comes from your hardware, not from LaserGRBL, and they are self-explanatory.
Wait if someone know Ortur laser better then me.

gmmanonymus111, 2020-05-06:
According to Ortur's manual, if your engraver is not in a job (or not moving) for 60 sec, or in weak mode, 100 sec, it'll turn off automatically for safety reasons. To resolve it, click on the `Lightning` icon in LaserGRBL, then the `Lock` icon.
Ortur engravers are using a different kind of microcontroller than arduino, and thus, a different version of GRBL. Since their code is not open source (at least I couldn't find it), it is impossible (very hard) to disable this feature. And since we mostly use arduino-based engravers, the compatibility of LaserGRBL with Ortur devices might be 'different'.

closed 2020-12-16 by arkypita