arkypita/LaserGRBL #970: Engraver creating a slanted image.

wladbg, 2020-04-24:
Hello! 
I have a problem with my 2.5w Chinese laser engraver: 
if I use Line to line horizontal tracking mode the engraver shifts each register creating a slanted image. 

[image]

If I use Vectrorize mode, looks much better: 

[image]

In Line to line - diagonal, is a disaster: 

[image]

LaserGRBL settings: 
10000 lines / mm, 
Speed: 1000 mm/min
s-min 0
S-max 500
Size: 43/10

I have the feeling that it is not working smoothly, or maybe something mechanical is wrong.

Video: [https://streamable.com/qoy9ga](url)

Do you have any idea what could be the problem?
Thank you!

StuartB4, 2020-04-24:
Have you just built the machine, if so are you sure the stepper motor leads are in the correct sockets on the main board.

StuartB4, 2020-04-24:
Can you post a picture / snapshot of your GRBL configuration settings.

wladbg, 2020-04-24:
Yes, it's my build. 
The mainboard is MKS DLC v2. 
All motors are connected exactly as in the manual, X-Y-Y. 
I have not changed anything in the configuration settings: 
[image]

StuartB4, 2020-04-24:
Start LaserGRBL and click connect. Then goto Grbl menu and select the grbl configuration like you did to get the list of settings. 
Change $32 to 1
I would lower $120 and 121 to about 2000.
Don't forget to click the Write button after changing the values.

Check to make sure none of the belts are too loose and that the toothed pulley wheels on the motor shafts are all tight.

Also use this online stepper motor calibrator to get the correct steps/mm.
https://jscalc.io/calc/tHjfIiKMKjDF42Hg?fbclid=IwAR219p6S0qWzVAsY8aLC8nyQaXaMpfa-mSTNCIXA7TAMPa3VzrGoCDcimsI

This is how to use it, but where it says inches select MM's instead.
[image]

wladbg, 2020-04-25:
After making all these adjustments, surprisingly it started working properly!
Only diagonally /line to line/ does not work, but horizontally and vertically it is OK now!
Thanks a lot for all your help, StuartB4!
Cheers!

closed 2020-05-06 by arkypita