arkypita/LaserGRBL #972: Loosing usb connection during job with laser always on

caigab, 2020-04-24:
Hi there,
I have a classic arduino+grbl 1.1 with a 20w laser on board.
Laser grbl 3.31 installed.
running the job when I unplug the alimentation of a  fan extractor  (on 220v)the machine stop and show me an alert popup about lost connections, the laser remain at the same power but the motor don't run so I experimented the joy of using a fire-extinguisher!
Fire and smoke outside my laboratory, nothing very exiting.
Anyway...this is the text received in the pop up:
LaserGrbl v3.3.1.0
Grbl v1.1h
Wrapper: UsbSerial
Win 7, Ultimate, 6.1.7601.65536, Service Pack 1 (0)

TypeOf exception  [System.ObjectDisposedException]
Exception message [L'handle Safe è stato chiuso]
Exception source  [System], thread []
Exception method  [Boolean GetOverlappedResult(Microsoft.Win32.SafeHandles.SafeFileHandle, System.Threading.NativeOverlapped*, Int32 ByRef, Boolean)]

   ----------- stack trace -----------
   in Microsoft.Win32.UnsafeNativeMethods.GetOverlappedResult(SafeFileHandle hFile, NativeOverlapped* lpOverlapped, Int32& lpNumberOfBytesTransferred, Boolean bWait)
   in System.IO.Ports.SerialStream.EventLoopRunner.WaitForCommEvent()
   in System.Threading.ThreadHelper.ThreadStart_Context(Object state)
   in System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   in System.Threading.ThreadHelper.ThreadStart()

I red the FAQ about the  USB cable  with ferrite that I do not have in this moment but I tryed 5 different cable without ferrite.
same issue,
Note: I've engraved many many hours before without facing this problem  so I need a Help to not incenerate my house.

arkypita, 2020-04-24:
Sei italiano?

arkypita, 2020-04-24:
Gli incidenti avvengono. Non lasciare la tua macchina incustodita mentre lavora!

LaserGRBL, che è un programma che gira sul PC, non può proprio fare nulla per impedire quello che è successo. Una volta che ha inviato i comandi alla macchina ha finito il suo lavoro.
Per di più se perde la connessione non può proprio fare nulla per dire al laser che deve spegnersi.

La macchina è indipendente dal PC. Se il PC si impalla, va in stand-by, si disconnette, o crasha, la macchina deve prevedere i suoi propri sistemi di sicurezza.

E' per questo che sistemi di sicurezza devono essere implementati a livello hardware nella macchina. Mi spiace, ma LaserGRBL non ha nulla a che fare con quello che ti è successo.

Sul tuo controller (la scheda che controlla laser e motori) gira un firmware che si chiama grbl che nasce per controllare le CNC.

Le CNC hanno problemi di sicurezza diversi dai laser. Se un CNC si ferma perché non riceve più comandi e i motori non si muovono, anche se il mandrino continua a girare non succede nulla. Su una marcatrice se il laser non si spegne succede quello che è successo a te.

E' proprio per questo motivo che a partire dalla versione 1.1 gli sviluppatori di grbl hanno aggiunto una flag di configurazione "Laser Mode" che prevede un funzionamento diverso per il laser tra cui quello di spegnersi automaticamente quando non c'è moto (se si attiva il LaserMode e si usa M4).

Ti suggerisco di leggere attentamente e fare le opportune considerazioni. https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode

caigab, 2020-04-26:
ti ringrazio infinitamente, userò sempre M4 per maggior sicurezza dato che ho grbl 1.1.

arkypita, 2020-04-28:
Per quanto riguarda il messaggio di eccezione ho cercato di approfondire.

LaserGRBL cerca di intercettare tutte le condizioni di errore previste senza mandare fuori questa finestra, che dovrebbe uscire solo in casi eccezionali.

Per esempio se si stacca il cavo, o se la porta smette di rispondere, non dovrebbe uscire questa finestra, ma LaserGRBL cerca di capirlo e dovrebbe semplicemente passare a stato disconnesso. Eventualmente, se questo accade durante la lavorazione, lo segnala con una finestrella come questa, che è la finestra che esce sul mio PC se stacco il cavo mentre sto facendo un lavoro.

[image]

La finestra che hai visto tu esce solamente quando succede qualcosa di imprevisto. 

Cercando su google sembra che solo su Windows 7 ci sia un problema con la porta seriale che fa uscire questo problema.

https://stackoverflow.com/questions/3808885/net-4-serial-port-objectdisposedexception-on-windows-7-only

caigab, 2020-04-29:
Intanto grazie ancora per il tempo dedicatomi.
Non mi sono spiegato bene probabilmente.
La schermata di errore che hai postato effettivamente è la stessa, ma il problema si presenta non se stacco il cavo usb ma la perdita di connessione avviene quando stacco la spina di qualsiasi utenza 220v io abbia collegato alla rete.
esempio:
sto lavorando con il laser ma mentre aspetto faccio un lavoro con una levigatrice o una taglierina che non centra nulla con il laser, stacco la spina dell'utensile per fare altro e perdo la connessione.
Pare un disturbo innescato dalla 220v oggi ho provato con un cavo usb con ferrite ma il problema rimane.
Ho controllato la versione di framework ed è la 4.0.
Proprio non me lo spiego.
Oggi provo con un pc win 10 per vedere se cambia qualcosa.

Di nuovo grazie se vorrai prendere in considerazione la mia mail.

Ciao

arkypita, 2020-05-01:
Ciao!
Si, avevo capito perfettamente.

Il tuo problema è sicuramente di tipo elettrico, un disturbo ti entra sul cavo e ti fa resettare l'arduino o qualcosa del genere. Verifica la messa a terra del tutto. 
Anche la scheda di controllo deve essere messa a terra, di solito questo avviene attraverso il cavo del PC ed il suo alimentatore, ma non è sempre così.

Se te la cavi un po' con l'elettronica prova ad aggiungere un condensatore elettrolitico da 1000uF circa sulla linea di alimentazione. Male non fa.

closed 2020-05-06 by arkypita

caigab, 2020-05-07:
Buongiorno Arkypita,
volevo aggiornarti su quello che ho potuto provare per risolvere:
cambiato il cavo con usb e ferrite
sostituito tutti i cavi dei motori, riguardato tutti i cablaggi
ho alimentato l'elettronica con una batteria a 12v separata dalla 220 e pilotato il grbl con un portatile scollegato dalla 220 running win x
ho messo un condensatore da 1000mF come consigliato
ho messo a terra la scheda con un cavo a parte
provato i settaggi su lasergrbl riguardo il buffering e il tread portandolo fino a "slow"
ho sostituito i driver A4988

Per ultimo ho tenuto stretto un santino di padre Pio per sicurezza....

niente da fare, qualsiasi cosa accada sulla rete 220 influisce sulla riuscita del lavoro.
Ora...grazie al comando M4 almeno non mi brucia la casa ma è veramente frustrante, sono dell'idea di cambiare l'elettronica  come ultima spiaggia, hai un consiglio da darmi su cosa scegliere?
Arduino Mega al posto di arduino Uno?
Precisazione, il tutto quando funziona lo fa perfettamente e ritengo anche di avere una discreta preparazione, ho costruito altre macchine come stampanti 3d e altro, non so dove sbattere la cabeza.

Grazie mille per il tuo impegno  ciao

StuartB4, 2020-05-07:
Ciao Caigab.

Ho appena letto il tuo ultimo post. Avevo una di queste schede che ho ottenuto con il mio incisore laser. È davvero semplice e fa bene il lavoro. L'ho usato per alcune settimane prima di venderlo perché avevo già uno scudo CNC Arduino Uno e Protoneer che avevo acquistato in precedenza.

https://www.amazon.co.uk/Woodworking-Engraver-Control-Controller-Engraving/dp/B07ZK9F1Z2/ref=sr_1_38?dchild=1&keywords=laser+cnc+controller&qid=1588852217&sr=8-38

Mi piace anche l'aspetto di questa tavola, sembra avere tutto il necessario, non troppo costoso. Non l'ho provato però. Ha microstepping per i motori. Ha anche 3 driver del motore passo-passo e può essere utilizzato come router / mandrino CNC se si desidera farlo. Anche connessioni per laser a 2,3 o 4 pin, in modo da poter aggiornare in un secondo momento, se necessario. È compatibile con GRBL 1.1.

[image]

arkypita, 2020-05-07:
Magari sei sfortunato e ti sei beccato un clone cinese di arduino bacato.
Ne hai altri da provare, magari di provenienza diversa?

Io suggerirei arduino uno originale, con protoneer CNC shield (anche clone).

arkypita, 2020-05-07:
Altro test che puoi fare è provare il tuo arduino staccato dalla scheda CNC, semplicemente collegato solo all' USB del PC, e vedere come risponde.

caigab, 2020-05-07:
Ho pensato anche a quello dato che la sfortuna non mi manca.
Appena possibile provo la soluzione dell arduino originale.
Grazie di tutto

Scarica Outlook per Android<https://aka.ms/ghei36>