arkypita/LaserGRBL #976: Improvement suggestion for user interface

frankvdh, 2020-04-26:
The separation of X & Y distance and direction controls is an irritation to me. I think it would be better to have a series of concentric circles, for each distance increment e.g. innermost would represent 0.1mm, next 0.5mm, 1mm, etc. This is not suggesting the circles themselves be those sizes... each circle would be the same width. The direction of movement would be indicated by where in the circle the mouse was clicked. The home button would be to one side, NOT be the innermost circle.

StuartB4, 2020-04-26:
Personal preference but it seems fine to me, and if it aint broke, don't fix it.
I also think it is easier as it is, because you set the amount you want it to move with the slider and then click the arrows to move it. 
If it's preset circles then it would be easy to click the wrong one.

frankvdh, 2020-04-27:
I'm new here, so I've just been putting up a few ideas that hit me, on the assumption that if it's non-obvious or counterintuitive to me, it would be the same for other newbies, and that the first things that hit you are likely the most important. If the consensus of the old hands is that it's good as it is, I'm happy to drop it.

StuartB4, 2020-04-27:
Maybe I'm a bit of a dinosaur and have got used to how it is.  I am getting on a bit now :)
I'm sure the developer Arkypita still appreciates new ideas/improvements.

arkypita, 2020-04-27:
I had implemented exactly what you are talking about in an old version.

[image]

I quickly revert this change because the lack of the "jog step" slider control (replaced by a control that simultaneously sets step and direction) makes it impossible to set the "jog step", which is used not only by the jog buttons but also potentially by the custom buttons and also also by the keyboard shortcuts.

Furthermore, if you think it could be used on touch-screen, the big button is certainly a more accessible thing.

Finally, I don't know how to handle it with the "continuous jog" mode which is one of the possible jog modes that can be set in LaserGRBL

closed 2020-05-06 by arkypita