arkypita/LaserGRBL #977: Vettorizzazione non incide tutti i punti

sylunder, 2020-04-27:
Ho due immagini con il nome delle mie nipotine.
Uso vettorizzazione ma l'incisione con il laser non risulta completa.
Mi puoi dare indicazioni per cortesia? Grazie
[Documents (2020-04-27 12-29-14).zip](https://github.com/arkypita/LaserGRBL/files/4539040/Documents.2020-04-27.12-29-14.zip)

StuartB4, 2020-04-27:
I nomi sembrano completi quando sono sullo schermo. Se lo fanno, potrebbe essere solo il legno che stai usando.

English:
Do the names look complete when they are on the screen. If they do then it could just be the wood you are using.

sylunder, 2020-04-27:
Ok per il legno ma il segno di bruciatura salta alcuni punti... Boh non so.

Uso Scala:"bordi netti".
Rimuovi macchie: 2
Arrotonda: 1
Ottimizza: 0,5
Clicco Optimize travel
Riempimento: nessuno
Spunto "adaptive quality"

Mi sai dire se devo modificare queste indicazioni quando uso la  vettorizzazione?

Grazie

> 
> 
> 
>     I nomi sembrano completi quando sono sullo schermo. Se lo fanno, potrebbe essere solo il legno che stai usando.
> 
>     English:
>     Do the names look complete when they are on the screen. If they do then it could just be the wood you are using.

_______________
Silvano Sottocasa
335 6315207
silvano.sottocasa@pec.it
sottocas@libero.it

StuartB4, 2020-04-27:
La tua macchina brucia i cerchi ok? Sono completi, perché le parti mancanti sono sulle curve.

Inoltre, puoi pubblicare una foto delle tue impostazioni GRBL. Puoi trovarli avviando LaserGRBL e facendo clic su Connetti, quindi fai clic sul menu Grbl e seleziona Configurazione Grbl. Dovresti ottenere un elenco delle impostazioni.

sylunder, 2020-04-28:
Ti allego gli screenshots delle impostazioni gbrl

Grazie

> La tua macchina brucia i cerchi ok? Sono completi, perché le parti mancanti sono sulle curve.
>
>     Inoltre, puoi pubblicare una foto delle tue impostazioni GRBL. Puoi trovarli avviando LaserGRBL e [...]

_______________
Silvano Sottocasa
335 6315207
silvano.sottocasa@pec.it
sottocas@libero.it

StuartB4, 2020-04-28:
Grazie ma non c'è nulla in allegato. Se lo stai facendo con il tuo telefono, non funzionerà. Devi farlo dal tuo computer.

sylunder, 2020-04-28:
Spero tu riesca a vederli...

> 
> 
> 
>     Grazie ma non c'è nulla in allegato. Se lo stai facendo con il tuo telefono, non funzionerà. Devi farlo dal tuo computer.

_______________
Silvano Sottocasa
335 6315207
silvano.sottocasa@pec.it
sottocas@libero.it

sylunder, 2020-04-28:
ti allego anche questo...

> 
> 
> 
>     Grazie ma non c'è nulla in allegato. Se lo stai facendo con il tuo telefono, non funzionerà. Devi farlo dal tuo computer.

_______________
Silvano Sottocasa
335 6315207
silvano.sottocasa@pec.it
sottocas@libero.it

sylunder, 2020-04-28:
Scusa se faccio più invii. ti allego le scritte da cui parto...

> 
> 
> 
>     Grazie ma non c'è nulla in allegato. Se lo stai facendo con il tuo telefono, non funzionerà. Devi farlo dal tuo computer.

_______________
Silvano Sottocasa
335 6315207
silvano.sottocasa@pec.it
sottocas@libero.it

arkypita, 2020-04-28:
Stuart intende che non puoi farlo rispondendo alla mail e mettendoli come allegat.
Devi andare sul sito di github e fare come hai fatto per lo zip del tuo primo messaggio.

arkypita, 2020-04-28:
QUI: https://github.com/arkypita/LaserGRBL/issues/977

sylunder, 2020-04-28:
scusami ma io ti ho sempre scritto per mail. come faccio a risponderti su #977?

> 
> 
> 
>     QUI: #977 https://github.com/arkypita/LaserGRBL/issues/977

_______________
Silvano Sottocasa
335 6315207
silvano.sottocasa@pec.it
sottocas@libero.it

sylunder, 2020-04-28:
Ti allego gli screenshots...
[image]
[image]
[image]
[image]
[image]

closed 2020-04-28 by sylunder

StuartB4, 2020-04-28:
Grazie. Ora posso vederli.

Puoi provare a cambiare $ 32 a 1 nelle impostazioni di configurazione di grbl. Avviare LaserGRBL e fare clic su Connetti, quindi fare clic sul menu Grbl e sulla configurazione Grbl. Modifica 0 su 1 nella casella del valore su $ 32. quindi fare clic sul pulsante Scrivi per memorizzare l'impostazione.

Controlla anche il menu Impostazioni LaserGRBL dal menu Grbl e controlla se Support PWM è spuntato. È la seconda scheda in alto. Se va bene, altrimenti selezionalo per abilitarlo. Quindi fai clic su Salva nell'angolo in basso a destra.

sylunder, 2020-04-29:
ti allego la foto prima e dopo le configurazioni di cui mi hai detto
[image]

StuartB4, 2020-04-29:
Sembra un po 'meglio.

Poiché il legno è di colore abbastanza chiaro, potrebbe essere un po 'più alto potere laser o una velocità leggermente più lenta.

sylunder, 2020-04-29:
hai un'ultima configurazione per  fare ciò? Giuro poi ti lascio in pace ;-))

StuartB4, 2020-04-29:
Haha. Solo modificando le impostazioni in Impostazioni immagine di destinazione.

Se stai usando M4 per accendere il laser, potresti provare a cambiarlo in M3. Solo per vedere se fa la differenza.

[image]

Inoltre, la regolazione di Luminosità e Contrasto può fare la differenza. A volte alcune parti dell'immagine scompaiono quando vengono impostate in modo errato. Come puoi vedere nelle 2 immagini sottostanti.
[image]

[image]

sylunder, 2020-04-29:
OK grazie. Risolto

StuartB4, 2020-04-29:
Buono a sapersi. Ora è tempo di divertirsi con esso. In bocca al lupo. :)

sylunder, 2020-10-26:
Ciao

uso laserGBRL alla grande. Ho una domanda da fare: se vedo che il lavoro non è come vorrei, fermo il programma... e cosa devo premere per tornare al punto "0"? Grazie

_______________
Silvano Sottocasa
335 6315207
silvano.sottocasa@pec.it
sottocas@libero.it