arkypita/LaserGRBL #978: ortur laser will not respond and has alrm code in corner

bulldog1358, 2020-04-27:
[image]
i have an issue with the laser not responding, and it has an alarm code in the corner. i could not get the alarm info to copy so i took a picture of it and attached it.. any help you can supply would be greatly appreciated. i just received this and have used it twice and now it is non functional.
thanks you mark beals

arkypita, 2020-04-28:
Alarm in corner is a message that is sent by the hardware. LaserGRBL only show it. 
Could be you have configured homing and hard limits but you don't have limit switch?
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#alarm-message

About exception message you report I am going to see causes and possible solution.

arkypita, 2020-04-28:
When you say alarm in corner you mean something like this?

[image]

bulldog1358, 2020-04-28:
Yes, it was in the corner. Not sure if this helps you or not but i shut everything down, laser and laptop. I then turned laptop on plugged in the laser and then turned it on. The alarm went away. Is there a sequence that one has to follow to turn on the laser for it to be seen on the laptop? 
I tried it again today and with the same procedure and it seems to be working if i turn it on in an certain order. 

Hope that helps.. 
thanks for responding.

Mark

gmmanonymus111, 2020-04-29:
When the alarm state pops up, you can try clicking on the '_Lightning_' button and on the '_Lock_' button in that order to reset and unlock your machine. This should put your machine in it's initial starting state. You should also re-do homing, as GRBL position might be lost

closed 2020-12-16 by arkypita