arkypita/LaserGRBL #979: Rotary axis

alssoliveira, 2020-04-27:
Hello developer, good afternoon!
I'm recording on a cylinder, for this, I used the Y motor cables and connected it to the rotary motor and it works well. However, the resolution is different for each diameter I am engraving.
Is there any way to create a text box where I can enter the value of the diameter and the software calculate the value of $ 101.
I even created a custom button to change the value $ 101, when recording on the X and Y plane the button inserts the default value, when using the rotary axis it changes the value regarding the resolution of the rotary axis. But the problem is that each time you change the diameter, the value of $ 101 also changes

Thank you and congratulations for the software. Simple and fantastic!

arkypita, 2020-05-06:
Tip: if you mount the object to be marked "in axis" with the motor (like on a lathe) you have the problem you are talking about, but if the motor connects it to a pair of roller, and on the rollers you place the object to be marked, the problem disappears!

[image]

ThomasSc, 2020-05-19:
> Hello developer, good afternoon!
> I'm recording on a cylinder, for this, I used the Y motor cables and connected it to the rotary motor and it works well. However, the resolution is different for [...]

I'm in the same situation.
Yes, a "Rotary Box" would be fine.
Thank you.

alssoliveira, 2020-05-19:
Thanks for the answer, but if we had the condition to inform the diameter to be recorded and he himself calculated the value of $ 101, it would be very good!
I found a way to use it and it works well ..
For the motor connection, I printed a key in 3D to change the motor from the Y axis to the rotary.
and the value of $ 101, I type in the instruction box: "$ 101 = (4000 / (3.14159 * Diameter to be recorded)"
and
"$ 101 = 80" to record on the Y-axis plane

photos attached

alssoliveira, 2020-05-19:
[image]
[image]

ThomasSc, 2020-05-19:
Nice.
I have made a custom button, but before switch steps, I have to calculate.

[image]

alssoliveira, 2020-05-19:
Great !.
If there is a field where it is possible to enter this value of 20 "(Ø = 20"), without having to edit the button code, it would be great !.

alssoliveira, 2020-05-22:
Laser engraving on rotary axis
https://youtu.be/VGTMlLLoROY

alssoliveira, 2020-05-22:
Laser engraving on rotary axis and plane surface
https://youtu.be/HUHdyw1YCl0

LMH33, 2023-03-23:
Can you help me how to calculate $101? I don't know how to calculate it, Thanks

[image]