arkypita/LaserGRBL #981: Swapping a new laser module in

LJg81, 2020-04-29:
Hi all

I realise this isn’t a GRBL-specific issue, but I’m new to the scene and this is the only laser community I’m in to date.

Had my laser working before I trashed it by running it at full power for too long. Have just received a new laser module to replace it.

The wiring is slightly different. I was after any advice anyone had for replacing the old 7kW laser module for this new 15kW one. 

Coming out of my Arduino(ish) board I have 3 wires, Laser 3-Pin labelled  Red ‘+’, White ‘s+’, Black ‘-‘.

On my new laser module there is a circuit board with a spare 2-pin port: labelled 
‘TTL (Analog) +’ 
and ‘TTL (Analog) -‘

Should I connect:

Arduino ‘+’ -> ‘TTL Analog +’
Arduino ‘s+’ -> nowhere
Arduino ‘-‘ -> ‘TTL Analog -‘

Asking for advice before I blow up another laser. Appreciate this is not the ideal place for this but new to lasers and GitHub and the scene.

Thanks in advance for any advice.

LJg81, 2020-04-29:
[image]
[image]
[image]

StuartB4, 2020-04-29:
The white and yellow connector from the laser with the board attached should go to the S and the - pins on the main arduino'ish controller board.  The red wire on the arduino board 3 pin connector is 12 volt power, but if you have a seperate power supply for the laser it is not needed. Just the S for the TTL/PWM signal and a Ground or -

StuartB4, 2020-04-29:
Just one thing, are the yellow and white leads marked on the laser board, so you know which is which.?

LJg81, 2020-04-29:
This is incredibly helpful @StuartB4 thank you.

I do have a basic wiring diagram with the new laser which does label the yellow and white leads thankfully, yes:

Yellow = TTL +
White = TTL -

Will update when I have rewired these connectors - thank you!

StuartB4, 2020-04-29:
Ok thanks.

LJg81, 2020-04-29:
@StuartB4 - it’s working like a dream. Thank you so much.

StuartB4, 2020-04-30:
That's great. :+1: 
Thanks for the update. Have fun.

closed 2020-04-30 by LJg81