arkypita/LaserGRBL #982: LaserGrbl v3.3.1.0

hans3254, 2020-04-29:
LaserGrbl v3.3.1.0
Grbl v
Wrapper: UsbSerial
Win 10, Professional, 10.0.18363.0 (7)

TypeOf exception  [System.ComponentModel.Win32Exception]
Exception message [Das System kann die angegebene Datei nicht finden]
Exception source  [System], thread [Main Thread]
Exception method  [Boolean StartWithShellExecuteEx(System.Diagnostics.ProcessStartInfo)]

   ----------- stack trace -----------
   bei System.Diagnostics.Process.StartWithShellExecuteEx(ProcessStartInfo startInfo)
   bei System.Diagnostics.Process.Start(ProcessStartInfo startInfo)
   bei LaserGRBL.MainForm.installCH340DriverToolStripMenuItem_Click(Object sender, EventArgs e) in E:\LaserGRBL\LaserGRBL\MainForm.cs:Zeile 567.
   bei System.Windows.Forms.ToolStripItem.RaiseEvent(Object key, EventArgs e)
   bei System.Windows.Forms.ToolStripMenuItem.OnClick(EventArgs e)
   bei System.Windows.Forms.ToolStripItem.HandleClick(EventArgs e)
   bei System.Windows.Forms.ToolStripItem.HandleMouseUp(MouseEventArgs e)
   bei System.Windows.Forms.ToolStrip.OnMouseUp(MouseEventArgs mea)
   bei System.Windows.Forms.ToolStripDropDown.OnMouseUp(MouseEventArgs mea)
   bei System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   bei System.Windows.Forms.Control.WndProc(Message& m)
   bei System.Windows.Forms.ToolStrip.WndProc(Message& m)
   bei System.Windows.Forms.ToolStripDropDown.WndProc(Message& m)
   bei System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   bei System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

gmmanonymus111, 2020-04-29:
It seems like you are trying to install the CH341 driver from LaserGRBL, but LaserGRBL can't find it in `{LaserGRBL directory}/Driver` folder. Am I correct?
If that's the case, place the content of this zip to `{LaserGRBL directory}/Driver`, then try again

[CH341SER.zip](https://github.com/arkypita/LaserGRBL/files/4554680/CH341SER.zip)

CJRN-cst, 2020-05-03:
Hello, kinda lost here. need to reverse directions on steppers, enable limit switches. Using Longruner GRBL CNC Shield Expansion Board UNO R3 Board. Have uploaded grblUpload.ino but cannot find where I can change configuration, do I have to create a custom ino file?

gmmanonymus111, 2020-05-03:
@CJRN-cst 
To invert stepper directions, you need to modify a GRBL setting (once you are connected to your board).
To do so, send `$3=6` from LaserGRBL's command field (or you can edit it directly at `File`->`GRBL Settings`-> `$3`). This will invert both X and Y axis.
More info about direction invert mask [here](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#3--direction-port-invert-mask).

To enable limit switches, you need to send `$21=1` (or again, you can edit it as mentioned above).
Wiring is the following (assuming you have a 3.0 CNC shield) X/Y limit pins and matching GND pins are correct, but Z limit switch (if you even use it) should go to Spindle Enable (SpnEn) pin. For 3.5 CNC shield, everything should be fine as it is.
For wiring limit switches, more info can be found [here](https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches).

CJRN-cst, 2020-05-04:
Thank you, got that to work. after all that, the module screwed up. Get what I paid for I guess.

arkypita, 2020-05-06:
> It seems like you are trying to install the CH341 driver from LaserGRBL, but LaserGRBL can't find it in `{LaserGRBL directory}/Driver` folder. Am I correct?
> If that's the case, place the content of this zip to `{LaserGRBL directory}/Driver`, then try again

That's strange! LaserGRBL installer copy CH341SER.EXE into C:\Program Files (x86)\LaserGRBL\Driver when the software is installed.

@hans3254 can you check if you have the file or not?

closed 2020-12-16 by arkypita