arkypita/LaserGRBL #984: Unsupported command

LJg81, 2020-04-30:
Hi, I wonder if anyone can advise.

I have several instances of ‘unsupported command’ when I view my Gcode in LaserGRBL.

I will attach a couple of photographs but I am wondering whether it explains some of the - mainly corner - parts of my design that the laser is not cutting (not coming on for).

Thanks in advance for any help

LJg81, 2020-04-30:
[image]
[image]

Have circled some areas in Red that have failed to cut. Wondering if the two are related. It looks like the unsupported commands seem to include 'G03'

arkypita, 2020-04-30:
This error is evidently an error generated by grbl, the program that runs on the control board. LaserGRBL simply brings it back to you. You should ask on the grbl support page. Not here.

It would seem caused by the exponential notation 1.2 .... E-12

What program did you use to generate the G-Code?

LJg81, 2020-04-30:
Thanks @arkypita - I'll post in the grbl page. Completely new laser cutting and hence these acronyms. Wasn't suggesting it was an error in LaserGRBL, just was on the look out for pointers. 

I used Scan2CAD to generate the G-Code.

Thank you, I'll look for the grbl page.

closed 2020-04-30 by LJg81