arkypita/LaserGRBL #985: Filling of vector image is not optimal

Jaisp, 2020-05-02:
Hi,

I wonder whether it is possible to optimize the filling of a vector image for laser engraving. If I would for example fill the image of a square, my laser will pass the whole square every time instead of filling each border seperately. Anyone with suggestions?

arkypita, 2020-05-06:
I have thought about this several times, but it is not easy to implement what you ask for.

gisep, 2020-05-18:
Would be easier to let recognize "closed regions" and fill them consequentially?

This wouldn't resolve this square's example but it could be done (for example) separating the drawing in two.

arkypita, 2020-05-18:
First of all you should know how LaserGRBL works.

The filling of the vectorized images is not done through vector calculation, but takes place by rasterizing the vector and applying the same code that is used in the "line2line" option to that raster image.

So to do what you propose it would be necessary to rasterize each path separately, apply the line2line algorithm, and accumulate the results by adding the movements necessary to move from one area to another.

Secondly, nothing guarantees that this work is actually more efficient. There are many cases in which passing the paths separately is much slower than doing them in a single pass. For example, in an image like this the time needed to mark the outermost path completely includes the time needed to make the interior.

[image]

The only way to get the best result is to translate each path filling into a series of vector operations, and process them all with a "Best Shortest Path" algorithm.

This is why I say "not easy to implement".

gisep, 2020-05-19:
Thanks arkypita for the explanation.

By the way, sometimes it could be convenient to fill the paths individually (for example when a few "black objects" are scattered in a large white space). Thanks to _evil mad scientist_ that work on EggBot and Axidraw, I've found a great extension for inkscape that can be really useful in these particular situations: this is called **Hatch Fill extension**.

WilsonandHowell, 2024-01-07:
The method I use is this:

1) I create my designs in whatever software will export a DXF file (Inkscape does this)

2) Open you DXF in QCAD (Freeware)

3) Select the outline and use the hatch tool in the left side toolbar (Icon looks like a tag with punchout. Click it then click the new single icon that looks the same, you'll see what I mean). Create whatever hatch when the dialog comes up and set line spacing values in the dialog box. (Inkscape's hatching methods are... never-mind)

4) Select the hatch (not the outline) and go to menu item Modify and down to Explode. Now the hatch is actual lines.

5) Save as same DXF file.

6) Open in Inkscape and save as SVG.

7) Bring into your laser software and it will be hatched/filled. The lower you set the spacing in the dialog the more fill you get.

Not sure if this is exactly what is being asked here but sounds like an answer to a common problem that sounds like this.