arkypita/LaserGRBL #986: lasergrbl laser working but not engraving

bdp137, 2020-05-02:
Everything seems to be working and engraving light is going over the board following the picture, but not actually engraving.

StuartB4, 2020-05-02:
Is it focussed properly? Are you trying to engrave on white material, that's harder to do because it reflects most of the laser power.
What are the power settings and speed settings, if it's too fast and not high enough power it won't work.
Have you managed to engrave anything at all before?

bdp137, 2020-05-02:
We just got this so we have never engraved before.  We know how to control the speed but are unsure of what the focus is?

StuartB4, 2020-05-03:
There should be an adjustable ring on the bottom of the laser, screw it in or out to get the smallest dot possible to focus it.

closed 2020-12-16 by arkypita