arkypita/LaserGRBL #987: custom buttons

Eme1942, 2020-05-02:
tried to install custombuttons after download. got message oops not your fault. would not let me copy text as you requested.

StuartB4, 2020-05-03:
Try this file. 4 buttons, these are the ones I use,
Frame prepared image.
Flash Laser.
Laser on for focussing.
Find center of prepared image.

Just download from the link below and import, no need to extract the file.
Just Right click on the buttons bar and click Import Custom Buttons and select the file you downloaded. It should ask you if you want to remove any buttons you may already have.
If you don't have any then you just click yes to the ones you want, you don't have to have all 4.

[CustomButtons.gz](https://github.com/arkypita/LaserGRBL/files/4569457/CustomButtons.gz)

Eme1942, 2020-05-03:
HI,THIS IS THE SAME MESSAGE I GOT WITH THE OTHER BUTTON  DOWNLOADS. ALSO WHEN I TRIED TO OPEN A FILE NOTHING HAPPENS. THE UNIT IS ON I CAN MOVE IT MANUALY BUT CANT OPEN ANY FILES.ALSO LASER BUTTON DISAPEARED. PLEASE HELP. ELLIOTTSent from Mail for Windows 10 From: StuartSent: Saturday, May 2, 2020 9:29 PMTo: arkypita/LaserGRBLCc: Eme1942; AuthorSubject: Re: [arkypita/LaserGRBL] custom buttons (#987) Try this file. 4 buttons, these are the ones I use,Frame prepared image.Flash Laser.Laser on for focussing.Find center of prepared image.Just download from the link below and import, no need to extract the file.Just Right click on the buttons bar and click Import Custom Buttons and select the file you downloaded. It should ask you if you want to remove any buttons you may already have.If you don't have any then you just click yes to the ones you want, you don't have to have all 4.CustomButtons.gz

StuartB4, 2020-05-03:
What do you mean the laser button dissapeared, you mean the Start button (Green Triangle)?
[image]

Eme1942, 2020-05-03:
> What do you mean the laser button dissapeared, you mean the Start button
> (Green Triangle)?

No the laser that was in the custom button to see the position.

>
> [image: iiiiiiiii]
> <https://user-images.githubusercontent.com/16980026/80919683-48bf5980-8d63-11ea-9a17-d0bf402958f2.jpg>
-- 
Elliott
eme1942@gmail.com
412 758 2484

StuartB4, 2020-05-03:
Have you tried downloading LaserGRBL again and reinstall it.

Eme1942, 2020-05-03:
I had an previous version   Problem started when I upgraded. I can’t find
last stable version. Could you send it to me to try. I really need to get
this working. I want to learn then move up a more substantial laser
engraver To start a business doing portraits.
Please help. Elliott

Eme1942, 2020-05-03:
I reinstalled the latest version 3 times!!!

Eme1942, 2020-05-03:
Are you in the U. S. Can I call you?

gmmanonymus111, 2020-05-03:
Could you please take a screenshot of the error message? Or try to copy it with `CTRL+C`? It would help a lot in solving your issue.

StuartB4, 2020-05-03:
This is the link for the latest version which I use and it seems to work fine.
https://github.com/arkypita/LaserGRBL/releases/latest/download/install.exe

LaserGRBL needs Microsoft .Net Framework software to work properly, so you could also update that, just to see if it makes a difference. The lastest version is here.
https://dotnet.microsoft.com/download/dotnet-framework

Also, you did'nt say which machine you have or the firmware it has.
Can you tell me that as well. The firmware version should show here when you start LaserGRBL and click Connect.
[image]

Eme1942, 2020-05-03:
Ortur master 15 watt
Grbl. 1.1f [‘$’ for help]
[MSG:’$H’|$x’  to unlock]’

Does this help?

Eme1942, 2020-05-03:
I sent screenshot of button problem. The file issue doesn’t give erro. When
I hit file- open nothing happens. I can take 2 screen shots but it won’t
show anything. It won’t open to select a file just like it won’t open to
install buttons.

gmmanonymus111, 2020-05-03:
Sorry, I can't seem to find any screenshots except Stuart's. Images sent by email are not appearing on GitHub. You need to post them directly here :/

Eme1942, 2020-05-03:
10

*From: *Stuart <notifications@github.com>
*Sent: *Saturday, May 2, 2020 9:29 PM
*To: *arkypita/LaserGRBL <LaserGRBL@noreply.github.com>
*Cc: *Eme1942 <eme1942@gmail.com>; Author <author@noreply.github.com>
*Subject: *Re: [arkypita/LaserGRBL] custom buttons (#987)

Try this file. 4 buttons, these are the ones I use,
Frame prepared image.
Flash Laser.
Laser on for focussing.
Find center of prepared image.

Just download from the link below and import, no need to extract the file.
Just Right click on the buttons bar and click Import Custom Buttons and
select the file you downloaded. It should ask you if you want to remove any
buttons you may already have.
If you don't have any then you just click yes to the ones you want, you
don't have to have all 4.

CustomButtons.gz
<https://github.com/arkypita/LaserGRBL/files/4569457/CustomButtons.gz>

Eme1942, 2020-05-03:
HERE IT IS AGAIN.

*ELLIOTT M. EDELSTEIN*
*MONROEVILLE, PA. 15146*
*412 758-2484 *

Eme1942, 2020-05-03:
Any idea why I can’t open file. That is more important

Elliott
eme1942@gmail.com
412 758 2484

StuartB4, 2020-05-03:
May be because it's an Ortur laser. A lot of people have problems with them.

Eme1942, 2020-05-03:
Can it be fixed. What can I do. It got good reviews on YouTube. That’s why
I got it

gmmanonymus111, 2020-05-03:
The laser might be of good quality, but uses a different kind of microcontroller and a 3rd party version of GRBL, which may cause problems with LaserGRBL compared to standard aruino based controllers.
I don't think the file opening issue is related to any engraving device. Can you open a file while your machine is disconnected?

Eme1942, 2020-05-03:
> The laser might be of good quality, but uses a different kind of
> microcontroller and a 3rd party version of GRBL, which may cause problems
> with LaserGRBL compared to standard aruino based controllers.
> I don't think the file opening issue is related to any engraving device.
> Can you open a file while your machine is disconnected?
>
> — I can’t open whether it’s connected or disconnected. I was able to do it
> before when I first connected it. But then I found the upgraded version of
> Laser Grbl. And it stopped working. Unfortunately I deleted the one that
> worked and now I can’t find it anymore. It’s sad Unfortunately I deleted
> the one that worked and now I can’t find it anymore. It’s said When I
> downloaded it that it was the last reliable I think it was that word
> version. I can’t find it on the Internet now. There was one place but it
> looks shaky asking for a donation right in the beginning which I would’ve
> done what the screen was blocked out so I couldn’t see what was underneath
> so I didn’t trust it. Is there anything I can do update my controller
can you refer me to a copy of the previous version. Please help me. I’m
trying to make a few bucks and I don’t have it too much money to lose. I’m
77 years old and  Right now I’m on a fixed income from the VA and Social
Security.

> As to hey third-party software the instructions were to load it downloaded
it from your site. That’s what I did. unless I was hijacked somewhere else
and didn’t know it..Please help me and tell me what to do

gmmanonymus111, 2020-05-03:
You can find all the releases, including older ones at https://github.com/arkypita/LaserGRBL/releases
If you can remember the version number, you can download that release, otherwise, you need to try them one-by-one

arkypita, 2020-05-03:
Hello, i am the developer under LaserGRBL

I don't know nothing about ortur laser. I have neither positive nor negative judgments.
As far as I know it is only a machine, like many others, that are sold on the internet.
If they distribute a customized version of LaserGRBL I can't help you.

Official LaserGRBL can be downloaded here: https://lasergrbl.com/download/

For a full reinstall you should:
1) remove current installation from control panel ([guide](https://www.laptopmag.com/articles/uninstall-programs-windows-10))
2) delete or rename LaserGRBL preference and settings folder located in C:\Users\YourName\AppData\Roaming\LaserGRBL
3) download and install LaserGRBL from official site: https://lasergrbl.com/download/

For step 2 please note that "AppData" folder is an hidden folder, 
To be sure to see that folder you should check "show hidden folder" in view options.

[image]

Now you have a clean, fresh install of official LaserGRBL.

Download my custom buttons from here: https://github.com/arkypita/LaserGRBL/files/4415289/MyButtons.gz

Or stuart ones, they are both good: https://github.com/arkypita/LaserGRBL/files/4569457/CustomButtons.gz

Right click on custom button area and select "import" option.
[image]

Select downloaded file and ask Yes to import dialog.

I am 100% sure that this procedure is working for official LaserGRBL.
If not, please attach video or photo of your error.

Eme1942, 2020-05-03:
Thank you so much for your help I have to leave now but I will be back in
the morning and I will try it. I will let you know how it works again thank
you so much for your trying to help me ELLIOTT

StuartB4, 2020-05-03:
@arkypita 

I know the Ortur uses a 32bit controller board, don't kow if it would cause all the problems though.

Eme1942, 2020-05-05:
Sorry I was sick yesterday. Did all you said. Same results. Just noticed when I was checking settings: There was no setting tab.I clicked in work area (see above). I noticed it is consistenly calling E:There is no E:    could that be the problem. It is looking fore the button and imiage files in e:IF THIS IS THE PROBLEM HOW DO I FIX. HOW DO I CORRECT LOCATION? ALSO WHERE IS SETTING TAB.THANKSELLIOTTSent from Mail for Windows 10 From: StuartSent: Sunday, May 3, 2020 7:29 PMTo: arkypita/LaserGRBLCc: Eme1942; AuthorSubject: Re: [arkypita/LaserGRBL] custom buttons (#987) @arkypitaI know the Ortur uses a 32bit controller board, don't kow if it would cause all the problems though.

arkypita, 2020-05-06:
I suggest you stopping using email to reply this thread.
Look like your email client is not compatible with github and in your reply we read a lot of weird text that make your messages difficult to read.

Simply go to https://github.com/arkypita/LaserGRBL/issues/987 and reply using reply box.
Thank you

Eme1942, 2020-05-06:
Sorry about that. I think I figured out the problem. I was able to drag a image in and engrave. When I tried to do the buttons I got that crazy message and I read it much slower this time. It looks like the problem is everything is being looked for on the E Drive. I don’t have an E drive. So that’s why I think we have the error. How can we change it? Also can you send me the text of the four buttons so I can enter it manually if necessary? Thank you ELLIOT

Eme1942, 2020-05-06:
Please acknowledge that you saw my last post. That could solve my problems I believe. Is there a way to get rid of that iDrive call. Also is it possible to get the text of the buttons so I can enter them manually please respond

arkypita, 2020-05-06:
I would gladly help you but your requests are not clear at all.
What are you trying to do when this "e drive" comes out? What is "e drive"?

LaserGRBL does not look for any "e drive" or "iDrive". Use the standard windows "open file" window. Use it to navigate the folders and choose the custom button file, as I described a few messages ago.

What do you mean you don't have the settings tab? The settings tab cannot disappear!

Put some photos. Describe it better.

Some custom button code can be found here:
https://github.com/arkypita/LaserGRBL/issues/306

Eme1942, 2020-05-06:
I’m having trouble sending you the picture. I am going to email my response please read it and get back to me. Error is showing That the program is looking for a drive that I don’t have. 

Elliott

Eme1942, 2020-05-06:
Here kit is please get back to me. I could not paste to your comment box.

Elliott

gmmanonymus111, 2020-05-07:
Sorry to tell you, but there are no images attached to the messages you were sending by email.
> **Images sent by email are not appearing on GitHub**. You need to post them directly here :/

Without images, or the complete error message, we don't know what have gone wrong
Try to take a screenshot of the error message by pressing the `Print Screen` key on your keyboard.
[image]
Then go [to this issue](https://github.com/arkypita/LaserGRBL/issues/987) with a web browser, then press `CTRL+V` (or right click and paste) to insert the screenshot here. When the upload is done, you can send the comment, and the image will apear here.

Eme1942, 2020-05-07:
https://share.icloud.com/photos/0epxDEeMDjkI4FXzdyRmBe_7g

Eme1942, 2020-05-07:
I sent you a link. You should be able to see that down. I could see the
image in my copy that I sent to you I don’t understand why you can’t see it
in the email. I had problems putting it on the web. I also found in this
image an error code 800401111

Please tell me the problem. Also I typed in some of the Buttons manually.
They still did not work. This is getting very frustrating. Who like the
program I’m trying to learn it but I need some help here please. Also can
you tell me if I want to write a G code program to cut and put a holes in
nylon strapping. Do I write the program and save it and drop that file into
the same Window that I would put images to engrave.

arkypita, 2020-05-07:
The image help me a lot.
For the future you can simply copy-paste the text from this error message.

The "E:\" drive path (E:\LaserGRBL\[...]) you can see in error message indicate the source code file that contains the faulting code. This path is refered to the PC where I have compiled LaserGRBL source code, so it has nothing to do with your PC, and it is not the cause of the issue.

The issue is an InteropServices.COMException in the system function "CreateVistaDialog".

Google about this issue "[COMException CreateVistaDialog](https://www.google.com/search?q=InteropServices.COMException+CreateVistaDialog)" gives some message that indicate that the issue is caused by an incompatibility between W10 visual theme and  high-contrast mode, when showing "open file" dialog.

Mothing to do with LaserGRBL itself, but i'll try to find out if I can put a workaround in LaserGRBL. I hope Microsoft will fix this issue with some updates!

I think that if you disable the high-contrast mode it should work.

arkypita, 2020-05-07:
Also check if your Windows is up-to-date with latest updates.

Eme1942, 2020-05-07:
My windows is up to date. Unfortunately I need the high contrast as I am
legally blind. I’ll try to turn it off to see if it works or not but I’m
pretty sure that won’t be able to see it too well.

Thanks

ELLIOTT

arkypita, 2020-05-07:
Yes, I imagined you had visual difficulties.

I have tried to make some changes in LaserGRBL that may have bypassed the problem. I released an experimental version to let you try them out.

You can download it here: https://github.com/arkypita/LaserGRBL/releases/download/v3.4.1/install.exe

I can't know if I solved it because on my computer the problem doesn't exist at all, so let me know if it solves.

Eme1942, 2020-05-07:
I turned off contrast and that worked except I couldn’t see what I was
doing too much. But I was able to load an image and set it up and engrave
it. I will download the new version   And try it was high contrast. I will
let you know. What is the smallest size that can be engraved. Also After I
downloaded and selected all of the speeds etc. the picture on the screen in
some cases was not clear. Especially when it came to a very small size such
as 17 mm x 20 mm

Eme1942, 2020-05-07:
It solves the problem. Thank you so much. Now I just have to learn all the
intricacies to adjust it for the different media that I plan to use. Have a
great day. Your customer service is the best day ever.  And I am not easy
to please as anyone that knows me would say. But you went above and beyond
and again thank you. ELLIOTT

arkypita, 2020-05-08:
You are welcome

> The picture on the screen in some cases was not clear. Especially when it came to a very small size such as 17 mm x 20 mm

For small images try using "adaptive quality" in vectorization mode:

[image]

wickeduf, 2022-02-04:
Hello, 

 I am having an issue with getting the custom buttons to download. I get an error message that says cannot open file"Mybuttons.gz". it does not appear to be a valid archive.

wickeduf, 2022-02-04:
I have windows 10 pro. I tried downloading .NET like someone suggested but that would not download either. I uninstalled and downloaded several times now.

arkypita, 2022-02-04:
[image]

gmmanonymus111, 2022-02-04:
@wickeduf You can't open the custom button archive, you need to import it to LaserGRBL.