arkypita/LaserGRBL #989: Laser Grbl crashing immediately

frankenkj, 2020-05-04:
I just installed Laser GRBL on a win7 laptop, and when launching the program after install it immediately crashes. I've uninstalled and tried previous versions, all on this machine do the same thing. I have Laser GRBL installed on a win10 machine without issues - are these releases windows specific ?
[image]

this is the error specifics:
Description:
  Stopped working

Problem signature:
  Problem Event Name:	CLR20r3
  Problem Signature 01:	lasergrbl.exe
  Problem Signature 02:	3.3.0.0
  Problem Signature 03:	5e98aa76
  Problem Signature 04:	mscorlib
  Problem Signature 05:	2.0.0.0
  Problem Signature 06:	4ca2b851
  Problem Signature 07:	f9d
  Problem Signature 08:	14e
  Problem Signature 09:	System.IO.FileNotFoundException
  OS Version:	6.1.7601.2.1.0.768.3
  Locale ID:	1033

gmmanonymus111, 2020-05-04:
It should work on Win7, as well as on XP
Do you have the correct .net framework installed on your Win7 machine?

arkypita, 2020-05-04:
.net framework is required to run LaserGRBL

Install Microsoft .net framework

frankenkj, 2020-05-04:
Just installed latest .net update, and I'm still getting the same error. Tried uninstalling and reinstalling laser grbl as well.

gmmanonymus111, 2020-05-05:
AFAIK, you need .Net 3.5 to run LaserGRBL. Newer versions of .Net are not backward-compatible with 3.5 (only from 4.5, but then XP support would be dropped)

frankenkj, 2020-05-05:
> 
> 
> AFAIK, you need .Net 3.5 to run LaserGRBL. Newer versions of .Net are not backward-compatible with 3.5 (only from 4.5, but then XP support would be dropped)

Thank you !
I uninstalled the .net latest version, reinstalled 3.5, and Laser GRBL Seems to be happy !

closed 2020-05-05 by frankenkj