arkypita/LaserGRBL #990: crashed after usb disconnect

sinfocomp, 2020-05-05:
LaserGrbl v3.2.0.0
Grbl v1.1 
Wrapper: UsbSerial
Win 10, Enterprise, 10.0.14393.0 (7)

TypeOf exception  [System.ObjectDisposedException]
Exception message [Se ha cerrado el controlador seguro]
Exception source  [mscorlib], thread []
Exception method  [IntPtr SafeHandleC2NHelper(System.Object, IntPtr)]

   ----------- stack trace -----------
   en System.StubHelpers.StubHelpers.SafeHandleC2NHelper(Object pThis, IntPtr pCleanupWorkList)
   en Microsoft.Win32.UnsafeNativeMethods.WaitCommEvent(SafeFileHandle hFile, Int32* lpEvtMask, NativeOverlapped* lpOverlapped)
   en System.IO.Ports.SerialStream.EventLoopRunner.WaitForCommEvent()
   en System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   en System.Threading.ThreadHelper.ThreadStart()

arkypita, 2020-05-05:
Hi! Thank you for reporting this exception.
It appears to have been caused by the .NET framework system serial driver, not directly by LaserGRBL. Unfortunately, since it is external code, it is difficult for me to catch this error.

I have already received a couple of reports, which seemed to limit the problem to PCs with Windows 7, but apparently there is also on your PC that has Windows 10.

I would really like to solve it, but the biggest difficulty is that on my PC I can't make it happen. Can you tell me if the error is sporadic, or does it always happen? Is there a particular procedure with which you can make it happen?

closed 2020-06-02 by arkypita

arkypita, 2021-01-01:
Hi
I write a new SerialUSB wrapper to comunicate with the engraver, that replace microsoft implementation that has some bug like the one reported here. 

If you want you can test by download from here: https://github.com/arkypita/LaserGRBL/releases/download/v3.8.10/install.exe and enable UsbSerial2 connection protocol (before connect!).

[image]

commit 035231f9 (2020-05-06): try fix #990
commit 52db15be (2020-06-02): SerialPort changes to fix #990 #1028