arkypita/LaserGRBL #991: Jog command fails at 4th quadrant

Bunbury-Berlin, 2020-05-05:
I'm a bit confused...

My machine- zero-position is at the upper left corner and I know that I have to shift all pictures to the 4th quadtrant by setting the pictures height as negative (Y) offset.

So my absolute working area is all positive X but negative Y only. Everything is working well, except jog commands. Every time I try to jog at the right-bottom-quadrant, a "software limit" failure occurs.

Now what's confusing me: I expected that Laser-GRBL does'nt calculate the software limits itself, but is listening to GRBL's-soft-limit-exceed-message instead. But other GRBL-clients (like candle) working well at the same machine, without any software limit violation... so it's a problem at Laser-GRBL only.

Any idea?

But it's a great software anyway, just like GRBL itself! 
Thank you guys, I've already donated and I will do again...

Bunbury-Berlin, 2020-06-03:
Supplement:  
jog command fails in -direction only if you're already in negative range. 

Example: 
machine zero is located in the upper-left corner, so my working area is 
X:  0 to +300
Y:  -200 to 0

Starting at 0/0 after G92 X0 Y0 command:

G00 X100 Y-100  works fine, but from this position, jog-buttons in Laser-GRBL works for ±X and +Y only, but negative -Y jog-button fails with range exceed message. 
(even if there's another -100mm travel range left and G00 X100 Y-120 would work properly)

arkypita, 2020-06-04:
I apologize for ignoring this report for too long. Free time is very few these days, I hope to be able to do tests as soon as possible and better understand the problem.

However thanks for reporting the issue

arkypita, 2020-06-04:
- Grbl version is 1.1 or 0.9?
- Are you using continuous jog or not?

[image]