arkypita/LaserGRBL #992: Cannot install drivers

frankenkj, 2020-05-06:
Win 10 x64 machine -

Laser Grbl installed and launches fine - when installing driver CH340 driver, it extracts and finds the inf, but errors out when installing. 
any ideas why ?
[image]

StuartB4, 2020-05-06:
You can try this version if you want. Just extract it and run it outside of LaserGRBL.
[CH341SER.ZIP](https://github.com/arkypita/LaserGRBL/files/4584383/CH341SER.ZIP)

frankenkj, 2020-05-06:
getting the same error

[image]

StuartB4, 2020-05-06:
Have you been in to windows Device manager to see if there is one already there, and a comport number.
[image]

arkypita, 2020-05-06:
Maybe you need administrator privilege?
Try right-click, run as administrator

frankenkj, 2020-05-06:
> 
> 
> Maybe you need administrator privilege?
> Try right-click, run as administrator

I tried that and same result.

Interestingly tho, without the dirvers installed I can control the laser via Laser GRBL and Lightburn.  Maybe it's  a non-issue ?

arkypita, 2020-05-06:
> Interestingly tho, without the dirvers installed I can control the laser via Laser GRBL and Lightburn. Maybe it's a non-issue ?

Maybe you have already installed them. Or you don't need them.

CH340 driver is needed only if your board use a ch340 chip for usb serial communication.
Different board types can use different chip.

closed 2020-12-16 by arkypita