arkypita/LaserGRBL #996: Manual Laser controll for focus

Schwarzi07, 2020-05-08:
My problem is that i can´t controll the laser by typing M3,M4,M5 or a S commnad with any value.
Am I doing something wrong or is LaserGRBL blocking direkt Laser controll?

gmmanonymus111, 2020-05-08:
That's how GRBL should work. You either have to type `G1 F1 M3 S1` (you need to 'move' to enable laser), or set lasermode to `0` then type `M3 S1` (and don't forget to set it back to `1` once done focusing).
My custom button for that:

- Button type: TwoStateButton
- Image: *The one and only you want*
- GCode: `$32=0 M3 S1`
- GCode2: `M5 S0 $32=1`
- Caption: *none*
- Tooltip: Turn on laser for focusing

Schwarzi07, 2020-05-08:
It worked!
Thank you for helping me.

closed 2020-05-08 by Schwarzi07