arkypita/LaserGRBL #999: Z axis homing not responding

armature81, 2020-05-11:
Hi Guys.
I have DIY laser cnc machine
I’m using arduino Uno and CNC shield with GRBL 1.1h and Lasergrbl software.

The issue is Z axis homing is not responding
Only y and x moving to homing as no movement from z.
Z axis I working if did jogging or even run gcode file.

Limit switches NC , hard wiring and limits pin are working fine for all axis as i tested.

I have checked config.h sketch and I can’t find any problem as shown below:

#define HOMING_CYCLE_0 (1<<Z_AXIS) // REQUIRED: First move Z to clear workspace.
#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS)) // OPTIONAL: Then move X,Y at the same time.
// #define HOMING_CYCLE_2 // OPTIONAL: Uncomment and add axes mask to enable

// NOTE: The following are two examples to setup homing for 2-axis machines.
// #define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS)) // NOT COMPATIBLE WITH COREXY: Homes both X-Y in one cycle.

// #define HOMING_CYCLE_0 (1<<X_AXIS) // COREXY COMPATIBLE: First home X
// #define HOMING_CYCLE_1 (1<<Y_AXIS) // COREXY COMPATIBLE: Then home Y

// Number of homing cycles performed after when the machine initially jogs to limit switches.
// This help in preventing overshoot and should improve repeatability. This value should be one or
// greater.
#define N_HOMING_LOCATE_CYCLE 1 // Integer (1-128)

// Enables single axis homing commands. $HX, $HY, and $HZ for X, Y, and Z-axis homing. The full homing
// cycle is still invoked by the $H command. This is disabled by default. It’s here only to address
// users that need to switch between a two-axis and three-axis machine. This is actually very rare.
// If you have a two-axis machine, DON’T USE THIS. Instead, just alter the homing cycle for two-axes.
// #define HOMING_SINGLE_AXIS_COMMANDS // Default disabled. Uncomment to enable.

// After homing, Grbl will set by default the entire machine space into negative space, as is typical
// for professional CNC machines, regardless of where the limit switches are located. Uncomment this
// define to force Grbl to always set the machine origin at the homed location despite switch orientation.
// #define HOMING_FORCE_SET_ORIGIN // Uncomment to enable.

So please any one can provide help or idea

StuartB4, 2020-05-11:
If you have the Z jog arrows showing can you actually jog the Z up and down?
[image]

armature81, 2020-05-12:
Yes i have and i can jog up and down also i can run gcode files and z axis is moving up and down 
Only homing is not responding

StuartB4, 2020-05-12:
Ok. Do you have any other old Grbl installations anywhere. It could be using the wrong config.h file with the Z axis homing commented out.

armature81, 2020-05-12:
I downloaded new GRbl 1.1h as I deleted the old

Schwarzi07, 2020-05-12:
In GRBL 1.1 the Spindle_Enable Pin and the Z_Axis_Limit Pin have been switched to add PWM support. You just need to plug your Laser/Spindle in the Z axis limit switch and your Limit switches in the Spindle_enable on the CNC shield

gmmanonymus111, 2020-05-12:
Silly question, but did you compile and upload the code to your arduino after editing the config.h file?
Also, make sure there are no more instances of `HOMING_CYCLE_0 ` enabled/defined later in that file.

armature81, 2020-05-12:
> In GRBL 1.1 the Spindle_Enable Pin and the Z_Axis_Limit Pin have been switched to add PWM support. You just need to plug your Laser/Spindle on the Z axis limit switch pins and your Limit switches in the Spindle_enable pins on the CNC shield

Yes I do following that pwm switched with z limit pin

armature81, 2020-05-12:
> Silly question, but did you compile and upload the code to your arduino after editing the config.h file?
> Also, make sure there are no more instances of `HOMING_CYCLE_0 ` enabled/defined later in that file.

Yes sure i did upload and homing cycle 0 is anabled

armature81, 2020-05-12:
[image]

$0 | Step pulse time | 10
-- | -- | --
$1 | Step idle delay | 25
$2 | Step pulse invert | 0
$3 | Step direction invert | 6
$4 | Invert step enable pin | 0
$5 | Invert limit pins | 1
$6 | Invert probe pin | 0
$10 | Status report options | 1
$11 | Junction deviation | 0.010
$12 | Arc tolerance | 0.002
$13 | Report in inches | 0
$20 | Soft limits enable | 0
$21 | Hard limits enable | 1
$22 | Homing cycle enable | 1
$23 | Homing direction invert | 7
$24 | Homing locate feed rate | 500.000
$25 | Homing search seek rate | 2000.000
$26 | Homing switch debounce delay | 250
$27 | Homing switch pull-off distance | 5.000
$30 | Maximum spindle speed | 256
$31 | Minimum spindle speed | 0
$32 | Laser-mode enable | 1
$100 | X-axis travel resolution | 160.000
$101 | Y-axis travel resolution | 160.000
$102 | Z-axis travel resolution | 708.000
$110 | X-axis maximum rate | 5000.000
$111 | Y-axis maximum rate | 5000.000
$112 | Z-axis maximum rate | 1000.000
$120 | X-axis acceleration | 3000.000
$121 | Y-axis acceleration | 3000.000
$122 | Z-axis acceleration | 500.000
$130 | X-axis maximum travel | 500.000
$131 | Y-axis maximum travel | 500.000
$132 | Z-axis maximum travel | 75.000

/*
  config.h - compile time configuration
  Part of Grbl

  Copyright (c) 2012-2016 Sungeun K. Jeon for Gnea Research LLC
  Copyright (c) 2009-2011 Simen Svale Skogsrud

  Grbl is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  Grbl is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with Grbl.  If not, see <http://www.gnu.org/licenses/>.
*/

// This file contains compile-time configurations for Grbl's internal system. For the most part,
// users will not need to directly modify these, but they are here for specific needs, i.e.
// performance tuning or adjusting to non-typical machines.

// IMPORTANT: Any changes here requires a full re-compiling of the source code to propagate them.

#ifndef config_h
#define config_h
#include "grbl.h" // For Arduino IDE compatibility.

// Define CPU pin map and default settings.
[... 632 lines cut ...]
// the spindle enable pin had to be moved and spindle direction pin deleted. The spindle
// enable pin now resides on A3, replacing coolant enable. Coolant enable is bumped over to
// pin A4. Spindle enable is used far more and this pin setup helps facilitate users to 
// integrate this feature without arguably too much work. 
// Variable spindle (i.e. laser mode) does NOT work with this shield as configured. While
// variable spindle technically can work with this shield, it requires too many changes for
// most user setups to accomodate. It would best be implemented by sharing all limit switches
// on pins D9/D10 (as [X1,Z]/[X2,Y] or [X,Y2]/[Y1,Z]), home each axis independently, and 
// updating lots of code to ensure everything is running correctly.
// #define DUAL_AXIS_CONFIG_CNC_SHIELD_CLONE  // Uncomment to select. Comment other configs.

/* ---------------------------------------------------------------------------------------
   OEM Single File Configuration Option

   Instructions: Paste the cpu_map and default setting definitions below without an enclosing
   #ifdef. Comment out the CPU_MAP_xxx and DEFAULT_xxx defines at the top of this file, and
   the compiler will ignore the contents of defaults.h and cpu_map.h and use the definitions
   below.
*/

// Paste CPU_MAP definitions here.

// Paste default settings definitions here.

#endif

gmmanonymus111, 2020-05-12:
Does your Z axis starts homing when your Z limit switch is disconnected?

armature81, 2020-05-12:
> Does your Z axis starts homing when your Z limit switch is disconnected?

No 
I tried many things like 
- change  S5 to 0 and remove all limit switches wiring... still not only z axis not moving.
- try  gbrl version 1.1e, 1.1f ... still same issue.
- finally test the machine with anther software ( Estllcam).. z axis homing working good.

gmmanonymus111, 2020-05-12:
Then I'm out of ideas... :/
LaserGRBL should send the `$H` command like (I suppose) EstlCam, and your MCU handles the rest in both cases.

armature81, 2020-05-12:
> Then I'm out of ideas... :/
> LaserGRBL should send the `$H` command like (I suppose) EstlCam, and your MCU handles the rest in both cases.

actually me too i'm surprise..!!!!
before it was working fine with grbl normally

armature81, 2020-05-12:
> Does your Z axis starts homing when your Z limit switch is disconnected?

No 
I tried many things like 
- change  S5 to 0 and remove all limit switches wiring... still not only z axis not moving 
- test the machine with anther software ( Estelcam).. z axis homing working good

> Then I'm out of ideas... :/
> LaserGRBL should send the `$H` command like (I suppose) EstlCam, and your MCU handles the rest in both cases.
> EDIT: One thing, you could try uncommenting `#define HOMING_SINGLE_AXIS_COMMANDS` and then sending `$HZ` to home Z axis with LaserGRBL. I'm curious wheter it works.

For single axis Homing  i tried to use it after i enabled from the config.h but no any axis respond to this commands.
I tried $HZ, $HY, and also $HX
I’m always getting errors

arkypita, 2020-05-12:
> LaserGRBL should send the $H command like (I suppose) EstlCam, and your MCU handles the rest in both cases.

Yes, LaserGRBL send $H
You can also try send it manually by typing $H in the "type gcode here" box, and hit enter.

[image]

> finally test the machine with anther software ( Estllcam).. z axis homing working good.

As long as elsecam sends $H the behavior must be identical, since the whole homing procedure is managed by the controller.

arkypita, 2020-05-12:
Are you sure you're not calling "homing" something that homing isn't?

StuartB4, 2020-05-12:
Is your cnc shield version 3 or 3.51. If it's 3,51 the Z limit pin and the SpnEn pins are correct when using 1.1 grbl. You only use the Z+/- pins for the PWM and the SpnEn pin for the Z limit switch on version 3 boards.

armature81, 2020-05-12:
> Are you sure you're not calling "homing" something that homing
[image]
only $H responding for Y&X but no action on Z
Single axis command not responding at all and u see in the picture there is an error

armature81, 2020-05-12:
> Is your cnc shield version 3 or 3.51. If it's 3,51 the Z limit pin and the SpnEn pins are correct when using 1.1 grbl. You only use the Z+/- pins for the PWM and the SpnEn pin for the Z limit switch on version 3 boards.

yes i'm using the correct pin for Z limit as i test the hardware with Estlcam as it works fine

armature81, 2020-05-12:
> Yes, LaserGRBL send $H
> You can also try send it manually by typing $H in the "type gcode here" box, and hit enter.
>
> [image]
>
>
> As long as elsecam sends $H the behavior must be identical, since the [...]

yes i did but same issue

armature81, 2020-05-12:
> yes i did but same issue

[image]
only $H responding for Y&X but no action on Z
Single axis command not responding at all and u see in the picture there is an error

arkypita, 2020-05-12:
> Are you sure you're not calling "homing" something that homing

[image]

Just to be extremely extremely certain that we are speaking the same language:

The button with the house positioned between the direction arrows is the jog to zero-zero button. We can call it "jog to home" button, and in LaserGRBL it send a jog command that move only X and Y "by design". The command sent is "$J=G90X0Y0F500"

The button with the house and the magnifying glass positioned in the toolbar is the Homing button, active only if homing is enabled in the grbl configuration and which controls the Homing feature.

Homing feature is the function of looking for limit switches by "touching" them to have an accurate and repeatable zero point each time it is turned on. This button sends the "$H" command.

The homing performance is ENTIRELY managed by the controller, so there can be no difference between how it is performed, whether you call it from LaserGRBL or run it from any other program.

arkypita, 2020-05-12:
Always assuming that the other program also sends the command $H and not some other strange command unknown to me :-)

armature81, 2020-05-12:
> > Are you sure you're not calling "homing" something that homing
> 
> [image]
> 
> Just to be extremely extremely certain that we are speaking the same language:
> 
> The button with the house positioned between the direction arrows is the jog to zero-zero button. We can call it "jog to home" or "return to home" button, and in LaserGRBL it send a jog command that move **only X and Y "by design"**. The command sent is "$J=G90X0Y0F500"
> 
> The button with the house and the magnifying glass positioned in the toolbar is the Homing button, active only if homing is enabled in the grbl configuration and which controls the Homing feature.
> 
> Homing feature is the function of looking for limit switches by "touching" them to have an accurate and repeatable zero point each time it is turned on. This button sends the "$H" command.
> 
> The homing performance is ENTIRELY managed by the controller, so there can be no difference between how it is performed, whether you call it from LaserGRBL or run it from any other program.

yes i'm using the correct button "  house and the magnifying glass"

armature81, 2020-05-12:
> Always assuming that the other program also sends the command $H and not some other strange command unknown to me :-)

that is way it's something strange for me

arkypita, 2020-05-12:
Do you know what command is sent by elekscam? Does it show it somewhere?

arkypita, 2020-05-12:
> For single axis Homing i tried to use it after i enabled from the config.h but no any axis respond to this commands.
> I tried $HZ, $HY, and also $HX
> I’m always getting errors

If you have enabled HOMING_SINGLE_AXIS_COMMANDS in config.h you should not have errors with $HX (I never tested this feature). Of course you should recompile and flash your arduino.

Are you sure you are editing the right config.h file? Something it may happen that you have multiple copies of the grbl source folder in different places.

armature81, 2020-05-12:
IT 
[image]
DOESN'T SHOW

arkypita, 2020-05-12:
What command did you send? XYZ?

armature81, 2020-05-12:
I tried $H and also $HX

armature81, 2020-05-12:
> If you have enabled HOMING_SINGLE_AXIS_COMMANDS in config.h you should not have errors with $HX (I never tested this feature). Of course you should recompile and flash your arduino.
>
> Are you sure [...]

i did delete fo all old copy and i downloaded new copy

arkypita, 2020-05-12:
I have doubts that the changes you make to the config.h file are not compiled.

Did you follow all the instruction in here: https://github.com/gnea/grbl/wiki/Compiling-Grbl
[image]

Can you try to flash a GRBL precompiled from LaserGRBL tools menu? I am sure that the 1.1h original (first item of menu) has Z homing enabled, and should home Z then X/Y.

[image]

armature81, 2020-05-12:
> I have doubts that the changes you make to the config.h file are not compiled.
>
> Did you follow all the instruction in here: https://github.com/gnea/grbl/wiki/Compiling-Grbl
> [image]
>
> Can you try to [...]

yes 
you are 100% correct
i used xloader and z axis home working fine

arkypita, 2020-05-12:
you are welcome

closed 2020-05-12 by arkypita