arkypita/LaserGRBL #1001: Terrible issue in gcode render

linewallker, 2020-05-12:
Hello, i am developing high resolution large formát grafical laser engraver and i am experiencing big error in gcode generated from pictures, There are strange lines which is definitely not part of the original pictures and every time I generated gcode they are in different position As you can see on attached pictures.
IT is a Very big problém for my projet and I need to know how to skip tru this problém. 

Thank you in advance for your quick answer. 

Petr
[image]
[image]

gmmanonymus111, 2020-05-12:
Could you tell more details about your machine and the settings used to generate the GCode (like lines/mm, dithering or line2line etc)?

linewallker, 2020-05-12:
I tried this ať the First time with experience on photos:
[image]
Now I change to this :
[image] 
IT will také maybe 2 days to plot enough to see how it is going

Machine is arduino vith cnc shield v3 and stepper driver a4988 on 1/16 stepp Ing right now and it is total reprap from old material
[image]

linewallker, 2020-05-12:
Also on smaller jobs
[image]

gmmanonymus111, 2020-05-12:
It seems a belt-driven machine, which might have issues when it comes to precise work. Although microstepping is meant to increase precision, it's not always working good. 1/16 steps are risky, because they can't hold a precise position (only whole steps are precise). Try to run it at 1/8 (that's what I use, but my machine is leadscrew-driven).
For the software part, make sure lasermode is enabled in GRBL (`$32` is set to 1) and you should use M4 for 'laser on command' in LaserGRBL while generating gcode. You could also try `1bit BW Dithering` instead of `Line-to-line Tracing`

linewallker, 2020-05-12:
On full step it was laso visible so I think that could be but it but on the other Hand if It is pressent on smaller jobs with full step setup it will not be the core problém, parts are havy so I think they should hold the position, also I mesuring the lenght every 1 Hour and both sides have the samé stepper lenght, also Distance between rows is looking Very good and quality of final plot is realy good, only those rasters which is occuring in grayscale is problem
[image]
[image]

As you can see details on bw plot is ok on full strenght of laser

linewallker, 2020-05-12:
Could it be caused with old grbl flashed in arduino? I think I am using old 0.9 flash

linewallker, 2020-05-12:
Also I will try s4 for laser instead of s3, I have not tried this option before

gmmanonymus111, 2020-05-12:
You need GRBL 1.1 to use [laser mode](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#32---laser-mode-boolean)
If you decide to upgrade, [this](https://github.com/gnea/grbl/wiki/Flashing-Grbl-to-an-Arduino) can be useful for flashing, or if you want to edit some advanced settings, take a look [here](https://github.com/gnea/grbl/wiki/Compiling-Grbl)

linewallker, 2020-05-12:
I have edited laser modulation speed in the files I have flashed, I increased to 10khz based on specifications from seller to get better performance, that is why I do nou upgrade, so I will upgrade but I must find how to Edit modulation speed in new version ☹️

StuartB4, 2020-05-12:
In Ggrbl 1.1 that can be done in the map.h file of grbl then recompiled and flashed to the arduino. 
But it  only goes up to 7.8 kHz.

>       // Prescaled, 8-bit Fast PWM mode.
      #define SPINDLE_TCCRA_INIT_MASK   ((1<<WGM20) | (1<<WGM21))  // Configures fast PWM mode.
      // #define SPINDLE_TCCRB_INIT_MASK   (1<<CS20)               // Disable prescaler -> 62.5kHz
      // #define SPINDLE_TCCRB_INIT_MASK   (1<<CS21)               // 1/8 prescaler -> 7.8kHz (Used in v0.9)
      // #define SPINDLE_TCCRB_INIT_MASK   ((1<<CS21) | (1<<CS20)) // 1/32 prescaler -> 1.96kHz
      #define SPINDLE_TCCRB_INIT_MASK      (1<<CS22)               // 1/64 prescaler -> 0.98kHz (J-tech laser)

linewallker, 2020-05-12:
That is a Very good tip which will savé a lot of gogling 😁 Thank you Very much for your time, now I have some useful tips and I am going to start some testing and let you know about final result 👍

StuartB4, 2020-05-12:
Ok. Good luck with it. :+1:

linewallker, 2020-05-12:
Thanks, I will get Back in few days, in this sizes it will be slow 😄😄😄

linewallker, 2020-05-13:
I think this probléms with raster lines occuring also on analog televisions and it is looking very similar to the lines on my plots, what do you think?
[image]
[image]
[image]

linewallker, 2020-05-17:
Hello again,
  I have updated to GRBL 1.1h onto device and also downloaded and instaled latest version of LASER GRBL 3.4.1 and alsu set up to M4 instead of M3 and the result is now without any raster but after 2 days of plotting I have experienced this 2 erroras at once and machine stopped and LASER GRBL froyed with 1.23 GB used RAM on about  1/3 of whole job.

ERRORS:

LaserGrbl v3.4.1.0
Grbl v1.1h
Wrapper: UsbSerial
Win 10, Professional, 10.0.19041.0 (0)

TypeOf exception  [System.OutOfMemoryException]
Exception message [Byla vyvolána výjimka typu System.OutOfMemoryException.]
Exception source  [mscorlib], thread []
Exception method  [System.Threading.ExecutionContext Capture(System.Threading.StackCrawlMark ByRef)]

   ----------- stack trace -----------
   v System.Threading.ExecutionContext.Capture(StackCrawlMark& stackMark)
   v System.Threading._ThreadPoolWaitCallback..ctor(WaitCallback waitCallback, Object state, Boolean compressStack, StackCrawlMark& stackMark)
   v System.Threading.ThreadPool.QueueUserWorkItemHelper(WaitCallback callBack, Object state, StackCrawlMark& stackMark, Boolean compressStack)
   v System.Threading.ThreadPool.QueueUserWorkItem(WaitCallback callBack, Object state)
   v System.IO.Ports.SerialStream.EventLoopRunner.CallEvents(Int32 nativeEvents)
   v System.IO.Ports.SerialStream.EventLoopRunner.WaitForCommEvent()
   v System.Threading.ThreadHelper.ThreadStart_Context(Object state)
   v System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   v System.Threading.ThreadHelper.ThreadStart()

Number 2 

sorry it fall down complety before I was able to coppy out … ( something about drawing or what )

But the problem with rastering is gone now, thank you for your good tips for finding the way :)

closed 2020-05-17 by linewallker

arkypita, 2020-05-18:
From your log i see that you have a 32bit system (32bit processor and/or 32bit OS)
In 32bit OS the address space available for application is limited to 2GB

LaserGRBL is very memory hungry because it has to work with bitmaps and lot of graphical objects.

I know it may not be that simple, but I suggest you use a 64bit processor PC if you need to work with such large images.

linewallker, 2020-05-19:
I did all your recomandations and also switched to x64 ultrabook with x64 win OS so I will see in few days if the probléms are completely gone.

Thanks a lot for your help and keep it going forward, it is cool software you are developing 👍😉