arkypita/LaserGRBL #1002: Arduino  + CNC shield Y axis problem

F10R1AIV, 2020-05-13:
Hi everyone,

I a having problems with the Y axis on my CNC. I am using LaserGRBL and the manual jog control works but once the cutting starts the y axis does not move anymore, any sugestion?

Thans in advanced.

arkypita, 2020-05-13:
Too high speed limit.

[Grbl Configuration ENG.pdf](https://github.com/arkypita/LaserGRBL/files/4621053/Grbl.Configuration.ENG.pdf)

F10R1AIV, 2020-05-13:
Thanks for the answer, i will try that and let you know, but it seems strange to my i have to modify this because i was running a lower soft version (3.1.0) and it was working fine, it was just after upgrading that starts failing.
Anyway i will tray to change the speeds. 

Thank you.

F10R1AIV, 2020-05-13:
Just modified speeds and accelerations but its the same. If you touch the Y axis motor pulleys they vibrate btw, but they dont move.

gmmanonymus111, 2020-05-13:
Could you try it with opening this .nc file (rename it to .nc from .txt)? It should draw a triangle without turning on the laser. (There are some invalid commands, please ignore them, they were used for testing error sounds)
[3szog.txt](https://github.com/arkypita/LaserGRBL/files/4623471/3szog.txt)

And could you show us the GCode generated by LaserGRBL?

F10R1AIV, 2020-05-13:
I just tested this triangle file and now the Y motors move, i am a little bit confused now, Why is that?

Thanks in advanced.

I insert link to the pictures.

https://ibb.co/6rkH7Ck
https://ibb.co/X8jP3j6
https://ibb.co/sVXRktn

gmmanonymus111, 2020-05-13:
By GCode generated I meant the one when you open a file in LaserGRBL.
In the picture, your feed rate seems a little bit high. That triangle uses a feed rate of 500. Your 3000-ish might be too high for the steppers. I had a similar issue,. Mine had a low acceleration, but once it reached a certain speed, it got stuck, vibrated and had a loud 'grrr' noise. After setting the maximum feed rate lower (several times, only by a little bit), it started working (I mean it did not stop working :P). I'm not sure about your acceleration settings, so I can't be sure about the solution.

F10R1AIV, 2020-05-13:
I again, sorry but i dont understand you about the generated Gcode, the link below is from the GRBL config setting i have.

https://ibb.co/nRFzWzX

Thanks

arkypita, 2020-05-13:
Y accelleration is too high. Why is different from X?
I already tell you the probable solution in my first message.

> it was just after upgrading that starts failing.

nothing in the update can cause this issue

arkypita, 2020-05-13:
Maybe also X and Y maximum speed. Try lower values until you find a working setup.

F10R1AIV, 2020-05-13:
I just changed the Y axxis acceleration because in the document you sant me before said that the setting the author was using was around 4000 as you can see in the attached link. The first setting i had was 1000, the same one as in the X axis.

https://ibb.co/j59h5v9

F10R1AIV, 2020-05-13:
I will try with lowering the speeds as you recomend. Thanks

arkypita, 2020-05-13:
They depends machine by machine.
Lot of variable plays in this values (stepper motor type, driver type, resolution, gear...) that is different for each hardware

F10R1AIV, 2020-05-13:
I just changed the speeds from 4000 to 1000 and then to 500 but the Y axis still dont move. Does the triangle file any different setting than normal?

arkypita, 2020-05-13:
A g-code file contains a series of command that move the laser, turn on/off laser etc
Trangle file start with M5 (turn laser off) than G0 (move fast to position x25 y46.3), than G1 move at controlled speed of 500 to X25.3Y45.7 than a series of movement using controlled speed (last G command was a G1) and using speed of 500.

Nothing strange in file.

```
M5 
G0X25Y46.3
G1X25.3Y45.7F500
X25.5Y45.5
X25.7Y45
X25.9Y44.8
X26Y44.5
X26.2Y44.3
X26.4Y43.8
X26.6Y43.6
X26.8Y43.1
```
Maybe you can save the file that gives you problem, using save (with advanced option) and upload here, so we can see what is the difference?

F10R1AIV, 2020-05-14:
Hi, good morning, i tried with two different files down below is the code from one of them

`G90 (use absolute coordinates)
M3 S0
F500
G0 X4.7 Y0 S0
G1 X4.75 S3
X4.8 S6
X4.85 S5
X4.9 S3
X4.95 S1
G0 X15.05 Y0 S0
G1 X15.1 S1
X15.15 S5
X15.2 S4
X15.25 S1
G0 X15.35 Y0.05 S0
G1 X15.3 S2
X15.25 S8
X15.2 S10
X15.15 S8
X15.1 S5
G0 X4.95 Y0.05 S0
G1 X4.9 S1
X4.85 S2
X4.8 S5
X4.7 S7
X4.65 S5
G0 X4.45 Y0.1 S0
G1 X4.6 S1
X4.65 S14
X4.7 S19
X4.75 S14
X4.8 S8
X4.85 S3
X4.9 S2
X4.95 S1
G0 X9.9 Y0.1 S0
G1 X10.1 S1
G0 X15.1 Y0.1 S0
G1 X15.15 S5
X15.2 S14
X15.3 S22
X15.35 S15
X15.4 S4
X15.6 S1
G0 X15.7 Y0.15 S0
G1 X15.65 S1
X15.6 S9
X15.55 S19
X15.5 S35
X15.45 S48
X15.4 S53
X15.35 S58
X15.3 S45
X15.25 S27
X15.2 S22
X15.15 S11
X15.1 S4
G0 X10.15 Y0.15 S0
G1 X10.1 S8
X10.05 S20
X10 S9
X9.95 S6
X9.9 S14
X9.85 S4
G0 X4.9 Y0.15 S0
G1 X4.8 S1
X4.75 S6
X4.7 S20
[... 6420 lines cut ...]
G1 X9.05 Y1.833
G1 X8.616 Y2.317
G1 X8.181 Y2.8
G1 X7.891 Y2.8
G1 X7.6 Y2.8
G1 X7.411 Y2.611
G1 X7.223 Y2.423
G1 X7.123 Y2.244
G1 X7.023 Y2.065
G1 X6.636 Y2.37
G1 X6.25 Y2.674
G1 X5.71 Y2.946
G1 X5.17 Y3.219
G1 X4.719 Y3.225
G1 X4.267 Y3.232
G1 X4.034 Y2.961
G1 X3.8 Y2.689
G1 X3.8 Y2.211
G1 X3.8 Y1.734
G1 X4.025 Y1.292
G1 X4.251 Y0.85
G1 X4.55 Y0.445
M5
G0 X0 Y0 Z0 (move back to origin)
`

F10R1AIV, 2020-05-15:
Good morning, did you find any difference between the gcodes?

arkypita, 2020-05-15:
nothing rilevant

F10R1AIV, 2020-05-15:
Any suggestion left to try maybe?

gmmanonymus111, 2020-05-15:
Your code works just fine. It contanins a lot of small movements, thus the 'tuktuktuk' sound for the X stepper (at least for me). Y moves rarely, and when it does, only moves less than a mm. Almost not noticable, but moves.
Coul you post your GRBL settings by typing `$$` or by navigating to `Grbl`->`Grbl Configuration`?

F10R1AIV, 2020-05-15:
Yes, the noise you mention looks like from the X-axis, down below you find the links to the pictures from the gcode.

https://ibb.co/cwF9GX7
https://ibb.co/dWWRfJD

gmmanonymus111, 2020-05-15:
It seems you sent the first part twice. The second post would be the most important.

F10R1AIV, 2020-05-15:
Sorry, you are right, here is the second part

https://ibb.co/WGWBWvW

gmmanonymus111, 2020-05-15:
Your max rate can be 2000 for both X/Y
Your acceleration for X/Y is high. Should be between 50-25.
If you have Z axis, the same applies to it regarding acceleration. Speed can be lower.

StuartB4, 2020-05-15:
Change $32 to 1 as well.

F10R1AIV, 2020-05-15:
Ok, i will try and let you know, very appreciated

F10R1AIV, 2020-05-16:
I made the changes all of you suggested and now it looks the machin can make a square without issues, ill keep trying different disigns but it looks good, very appreciated for the help.

closed 2020-12-16 by arkypita