arkypita/LaserGRBL #1004: Request: Cut order options

MarkDowes, 2020-05-15:
Hi, not sure if this is the right place for a request, delete/ move if not. Anyway.

I am using LaserGRBL for cutting paper/card with vector input.
Usually, the order it decides to cut lines in is all over the place, a lot of time wasted moving back and forth a long distance to cuts rather than moving on to the next closest, etc.

If at all possible, some options for re-arranging the cut order would be great for me. Some of the options I have seen in competing software include choosing a corner/ side to work from and it moves across from that starting point, and cutting inner lines first. Things like this would be very useful for me.

Thanks for your time and for the great software!

rchouinard, 2020-08-26:
I second this request. I sometimes have to cut thin materials, and it would be much better to cut the inner-most features before the outer-most to avoid shifts due to material curling or breaking free. For example, cutting letters like "O." It would be much better in some cases to re-order the cut to the inside circle first,

Rybec, 2021-05-16:
If you are using Inkscape, this is possible by changing the render order of the paths.  If you have objects with multiple paths, you will have to convert them to paths (in menus Path -> Object to Path, with the object selected), and then break them into independent paths (Path -> Break Apart).  From there, select the paths you want to cut last, and move them to the bottom (Object -> Lower to Bottom).

The problem here is that there are cases where it is important to cut the outside features first and then the inner features.  (I've come across a few personally, on a few occasions.)  This isn't as common as the reverse, but if LaserGRBL automatically changes the order of cuts, it can create serious problems where specific cut orders are needed.  (Also, the sort of collision detection required to handle complex objects is extremely complicated.  You can't just use bounding boxes.)  Thus it is better to let the user determine cut order.  Not all CAD software will let you do this, but if you can export in SVG, you can use Inkscape to reorder elements to get the cut order you need.  (In theory, you could fully order your cuts, by selecting paths one by one, and using Object -> Raise to Top, in the order you want them cut.  For very complex parts though, this is not really feasible.)

Do note, however, that Adobe is not SVG standard compliant.  It's doesn't use the scale specified in the SVG standard, so exporting to SVG will get you the wrong size.  If this is an issue for you, I suggest switching to _professional_ drafting software, made by people who actually care about things like usability.  I just use Inkscape for 2D drafting, and it works very well.  Alternatively, I am sure if enough of Adobe's customers raised a stink about this, they would eventually fix it.

tistructor, 2022-01-28:
I found this plugon for inkscape which seems to work fine.
It optimized my svg file perfectly.
https://wiki.evilmadscientist.com/Installing_software

This is the screenshot of the optimized file:
[image]
The non-optimized version I created with Kicad took over an hour to get the labels.

tistructor, 2022-01-28:
It could save you additional time if the cutting direction of the line segments also changed.
Even if you can't really complain.

sebastianheyn, 2022-11-01:
> If you are using Inkscape, this is possible by changing the render order of the paths. If you have objects with multiple paths, you will have to convert them to paths (in menus Path -> Object to [...]

I tried this procedure but it doesnt work, lasergrbl will always print the outline first, no matter how low I put the outline path - I have also tried to create a second layer, and lowered it, but same result. If I open the file with lightburn, it renders the lines in the order they are stored in the svg.

bcjams, 2022-11-24:
I know this is an older thread but I just started using Lasergrbl, and I had same issue.  What I found is that reordering Inkscape does work, but the trick is after reordering, you need to recombine the paths by using Path>Combine

benistote, 2023-03-31:
I have a complex file with many curved paths and the EggBot extension didn't seem to work well for me... If anything, it seems even worse now. I don't know why it did that.
Concerning the inner/outer lines problem (that I also had because I need to cut shapes with holes in them, what I did is that I made my file as normal, then I saved two versions of it: version A is the inner lines (I selected all the outer lines, which for me were regular circles so it wasn't too time consuming, and deleted them), and version B is the outer lines (I selected the inner lines and deleted them).
So you get two files, and in LaserGRBL you go to "File->Open a file" and you open version A, then you go to "File->Add a file", and you add version B. The laser will now cut all the lines in version A (inner lines) and only cut version B (outer lines) after the prior file has been completed.
Hope that helps.