arkypita/LaserGRBL #1007: Can't laser grayscale don't have M4 or $$ 32

vasjo1, 2020-05-17:
Hello, I watched your movie but I still can't get out of my problems.
I cannot select M4 anywhere to laser the gray scale in a photo, with me all lasing at the same strength. Also the $$ 32 I do not get in my settings but it goes to $$ 27. Have already reinstalled but it remains the same.
The program on the GRBL site under GRBL v1.1 is also no longer available.
Can you tell me how I solve this problem.
And my laser does not make a smooth movement when lasing, it goes shockingly to the left and to the right, only the rapid traverse goes smoothly.
Sorry for the bad English
Greetings from Belgium
Jozef

StuartB4, 2020-05-17:
You can upgrade to GRBL 1.1 in LaserGRBL. It's on the Tools menu,  Flash Grbl Firmware.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

vasjo1, 2020-05-19:
Hello, 
thanks for the tips on how to install it, and now I can indeed print the photos as it should. 

Thanks and greetings from Belgium. 
Joseph 

[ https://user-images.githubusercontent.com/16980026/82161937-bbb6ed00-9898-11ea-8edb-64ea47cb0fa4.jpg ] 
[ https://user-images.githubusercontent.com/16980026/82161938-bc4f8380-9898-11ea-9720-70ae8d97f09f.jpg ] 
[ https://user-images.githubusercontent.com/16980026/82161939-bce81a00-9898-11ea-8765-05ccf6ffd872.jpg ] 
[ https://user-images.githubusercontent.com/16980026/82161940-bce81a00-9898-11ea-84e2-f9ebab5b8d73.jpg ] 
[ https://user-images.githubusercontent.com/16980026/82161941-bd80b080-9898-11ea-886a-1cd384899c09.jpg ] 
[ https://user-images.githubusercontent.com/16980026/82161943-bd80b080-9898-11ea-979e-d177438b6041.jpg ] 
[ https://user-images.githubusercontent.com/16980026/82161944-be194700-9898-11ea-8fbf-cc2da2cef072.jpg ]

StuartB4, 2020-05-19:
You're welcome. :+1:

closed 2020-12-16 by arkypita