arkypita/LaserGRBL #1008: Laser works on full power all the time.

WDRplnl, 2020-05-18:
Hi.
I've connected my laser to shield&power and it's working only on 100% power after making switch ON on laser.  All on foto.
[image]
Any suggestions to solve that dear PRO's ?

StuartB4, 2020-05-19:
If you have the yellow/brown PWM wire correctly connected to the Z+ or Z- pin then everything looks as it should.
It does'nt matter which Z pin because they are both connected together on the underside of the shield on the circuit. 

I've had mine come on full power sometimes and then realised the PWM wire just was'nt making a good connection on the Z- pin. 

You could also check on the underside of the shield for a bad connection on the Z pins, and where the yellow and brown wires are joined.

The only other thing I can think is that with some laser driver boards when the PWM signal is 0v you get full power and when it's 5v it's off.  It's opposite to the normal way of doing it. But without knowing the make or supplier of the laser it's hard to tell if that one is like that.

Schwarzi07, 2020-05-19:
The Laser looks like a Eleksmaker replacement laser, but they all use the same driver so it isn´t possible to find out what laser that is.

WDRplnl, 2020-05-19:
I'm thinking about way to invert signal. Mosfet PWM switch wont work that way?
[image]

closed 2020-05-19 by WDRplnl

StuartB4, 2020-05-19:
That's good. :+1: