arkypita/LaserGRBL #1009: laser not functioning properly

Clarky00, 2020-05-19:
the laser isnt functioning properly. only had it 3 weeks . it worked at the beggining but now not fully engraving properly.

Schwarzi07, 2020-05-20:
Please provide more Information.

Clarky00, 2020-05-20:
Hi yes I will send images of what it’s doing . [Image.jpeg]

gmmanonymus111, 2020-05-20:
Sorry to tell you, but images sent by email don't appear on GitHub. Please, post them directly here.

Clarky00, 2020-05-20:
[Image.jpeg]

Clarky00, 2020-05-20:
Has this worked ?

gmmanonymus111, 2020-05-20:
`[Image.jpeg]` showing up, but there is no image :/

Clarky00, 2020-05-20:
Ok I will try another way . Basically the laser only does half an image . It tried to do it but not consistent if that makes sense . ? If not could someone call me ?

gmmanonymus111, 2020-05-20:
Without laser and/or engravery type, I'm guessing...
Does it say `Alarm` or `Pause` at the lower right corner when it stops?
[image]
Or does it continue moving without the laser being powered on?

Clarky00, 2020-05-20:
It’s dosnt stop .

[image]<https://user-images.githubusercontent.com/9658243/82458406-eaaaaa00-9ab6-11ea-9969-570ba65ab361.png>

gmmanonymus111, 2020-05-20:
Does it stop engraving at the same spot, or at randomly?
It might worth checking your TTL wiring, as it might come loose, rendering your laser 'always off'

Clarky00, 2020-05-20:
Randomly .

Clarky00, 2020-05-20:
Wiring checked.

Clarky00, 2020-05-21:
[Image.jpeg]This is what it’s doing . We’ve tried some today on some it’s fine then it will just start doing this ! .

gmmanonymus111, 2020-05-21:
Images sent by email are not showing up.
[image]
Please paste them directly at the [issue](https://github.com/arkypita/LaserGRBL/issues/1009)

Clarky00, 2020-05-21:
[image]
[image]

gmmanonymus111, 2020-05-21:
What is your laser's optical power? It seems the laser fails to start lasing as the material is too light-coloured. Similar when you try to cut something from white paper, the laser sometimes does not have enough time to 'bite' in the paper, and can't start cutting.
Try lowering engraving speed or giving more power to your laser durnig your engraving job.
Or if you can't adjust those values, try darker wood.

Clarky00, 2020-05-21:
Engraving speed 7800
S max to 585 .
We’ve printed on the same material for the last few days and it will be ok one min then not another .

StuartB4, 2020-05-21:
That's a fast engraving speed, Slow it down a bit.

Clarky00, 2020-05-21:
[image]
[image]
We has changed the speed (faster and slower) we have also changed the s-max power up and down but nothing seems to be working and we have been using the same wood up to now with no problems.

StuartB4, 2020-05-21:
It also looks like it might be slightly out of focus, have you tried adjusting the lens to get a smaller sharper dot. The laser dot is a rectangle not a round dot. So you really have to get it as small as possible. If you can see it with eye protection on then focus it you will notice that the sides of the rectangle start to move in to the center, if you go too far the top and bottom lines of the rectangle start to move apart.

Karmael19, 2020-05-24:
HI,
We now have the same problem. It was a "small circle" and now it is this rectangle shape, I have tried to make it go back to the small circle without luck, any ideas?

StuartB4, 2020-05-24:
When you focus it use a black surface, you can see the dot better. I'ts only a really small amount between in and out of focus.

Karmael19, 2020-05-24:
Thank you for that. We ended up taking apart the laser completely and re-doing it. We found a few loose screws (lol not us mind you) and were able to re-focus the laser. If it happens again I will do what you suggested !

StuartB4, 2020-05-24:
Ok. :+1:

Clarky00, 2020-05-24:
Thank you we will have a look .

Ok. 👍

closed 2020-12-16 by arkypita